matthijskooijman/arduino-lmic #5: Arduino lmic problem to run sample code.

po4life, 2015-09-16:
Best @matthijskooijman 

Sorry for posting my previous issue on the wrong spot!

But we have the sx1272 and when I debug the running code the value of v is 255. I thought this is a really weird value to get back. If the SPI connection isn't set up properly, how do i fix that of what can the problem be?

Greetings Po4life

matthijskooijman, 2015-09-16:
255 sounds like a bad SPI connection, yes, where it's just reading ones. Did you connect MISO/MOSI/SCK to the corresponding pins on the Arduino? Did you connect the SS (slave select, also called chip select) to an Arduino I/O pin and set the right pin number for it in the pin mapping?

po4life, 2015-09-16:
I didn't changed anything in the pin mapping, but thanks for the information!
Just readed that you runned this test program on an other adruino platform.
I will contact you when it's succeed or not.

po4life, 2015-09-16:
I changed the pins, but the rxtx are on different ports. The error is changed to line 513 in radio.c and with the serial monitor I see that the startup works. Now he goes into the loop and gives a failure back after running the os_runloop();.

Any idea if this has connection with the rx and tx that are separated from each other or that it is something else?

Greetings Po4life

matthijskooijman, 2015-09-17:
What board do you have exactly? The rxtx pin mentioned here is a pin to switch the antenna between the Receiver and Transmitter pins on the SX chip. If you have a separate RX and TX pin, that suggests that you're actually using a module that does not allow direct access to the SX chip, but instead offers a serial interface to be used. If so, this library is not applicable.

po4life, 2015-09-18:
We are using the arduino uno with a cape on it. This cape is the multi protocol radio shield of cookinghacks and then we set the libellium rf board with sx1272 on the socket 1. That is our setup now. 
But the pin mapping that you mentioned earlier are the configuration of the arduino board or the rf board?

matthijskooijman, 2016-02-20:
Did you manage to solve your problem? I've just pushed a major update of this library, which includes much improved documentation. I'll go ahead and close this issue, but feel free to add more comments so we can reopen it if you're still having problems.

closed 2016-02-20 by matthijskooijman