matthijskooijman/arduino-lmic #10: RE:arduino-lmic

glennsibs, 2016-03-04:
hie there I am trying to build a Lora node to node communication using two mini arduinos and twoFM LORA TRANSCEIVER MODULE, PRE SET TO 868MHZ which has an SX1272 Chip, im having sme errors on my arduino, its not compiling instead im getting these errors...
\ Build options changed, rebuilding all
C:\Users\x00097124\Documents\Arduino\libraries\arduino-lmic-master\src\hal\hal.cpp: In function 'u4_t hal_ticks()':
C:\Users\x00097124\Documents\Arduino\libraries\arduino-lmic-master\src\hal\hal.cpp:154:115: error: 'static_assert' was not declared in this scope
static_assert(US_PER_OSTICK_EXPONENT > 0 && US_PER_OSTICK_EXPONENT < 8, "Invalid US_PER_OSTICK_EXPONENT value");
^
Error compiling. //
what could be my problem, please help...

matthijskooijman, 2016-03-07:
As I replied by e-mail as well, you're probably using an older version of the IDE, or a third-party board/core that is missing C++11 support. Also, please refrain from adding comments to all kinds of unrelated issues and commits, that will only cause confusion. Let's stick the discussion about this problem in this issue, so let me know if the above helps you resolve it.

matthijskooijman, 2016-03-10:
You've replied by e-mail that it is working now, so I'm closing this issue.

closed 2016-03-10 by matthijskooijman

ahmed9207, 2017-04-05:
@glennsibs could you share how you fixed the problem please , i'm having the same compilation error.
thanks

matthijskooijman, 2017-04-05:
@ahmed9207, I suspect that you are using a board which does not have C++11 enabled in the compilation options. To fix that, you would have to change `platform.txt` to include the `std=c++11` option in the right places (or perhaps update your board file and/or contact the board/platform maintainer to ask them to).

ahmed9207, 2017-04-05:
Ok, thanks for the quick reply .