matthijskooijman/arduino-lmic #9: SPI clock issue on ATmega328

Oliv4945, 2016-03-04:
Hi,

Hi tried your library with an RFM95 chip and the "raw" exemple, and it works with an ATMega1280 but not 328... I hooked up an oscilloscope and spotted that there is only 3 clock cycles !
I saw on TheThinkNetwork that you tried it on ATMega328, did you already run into this issue ?

Thanks !

matthijskooijman, 2016-03-07:
It worked for me using a 328. Also, the hardware SPI unit is used through the standard Arduino interface, so I would expect that to work.

Got any images of the scope view? What board are you using exactly?

Oliv4945, 2016-03-07:
I know that SPI communications are done by Arduino libraries, that is surprising to me too...
I tried with an official Arduino Uno and a cheap clone, both have same result. Arduino software is 1.6.7

Hereunder is the oscilloscope capture.
* Yellow : clock
* Green : slave select (3.3V offset)
* Blue : MOSI

[image]

matthijskooijman, 2016-03-10:
Weird. What happens if you disconnect the SCK line? Is anything else connected to the board? What happens if you use a simple SPI example sketch?

Oliv4945, 2016-03-10:
Funny, I just tried without anything connected to the Arduino and the example sketch, and I have the same result. I moved to an Arduino Nano, I only have one clock cycle.
On the Mega1280 it is ok, so I will close this issue and ask Arduino guys, but could you please tell me which Arduino version are you using ?

Thank you, and I let you know what happened

closed 2016-03-10 by Oliv4945

matthijskooijman, 2016-03-10:
I'm using some git version of Arduino, probably a bit newer than 1.6.8.

Sounds like something weird is going on - I would doublecheck your test setup. Or perhaps you've miswired something previously that caused the SCK pin on all of these Arduinos to become damaged?

Oliv4945, 2016-03-10:
I think my SPI port is OK at least on the Nano as I use it as an ISP programmer, also one of the Arduino Uno I tried was brand new and official...

On the test setup I used both examples "barometricPressureSensor" and "digitalPotControl" but I will try to write a simple sketch to be able to test all SPI modes before reporting the issue

Oliv4945, 2016-03-15:
Finally it works... but that was a weird issue !

First, the laptop I used is dying and the 5V was not stable => reset during RFM initialization.
But with Arduino r3 the clock speed you defined (10E6) is not reachable, when I hook in an oscilloscope the slew rate prevent to have full clock signal. I tested it with success with 2E6.

By the way, thanks for the library !

matthijskooijman, 2016-03-16:
@Oliv4945, interesting, thanks for reporting back. Regarding the clock speed, 10E6 is just what the chip supports, IIRC the Atmega only goes up to F_CPU / 2. Apparently your hardware setup prevents you from reaching even that, so adjusting it seems reasonable.

Oliv4945, 2016-03-17:
Oh, true, I forgot the prescaler :/
Nevertheless I tried SPI@8 MHz with the Uno r3 without anything connected and it is not able to reach such speed.