matthijskooijman/arduino-lmic #13: Wrong DIO function mapping

msamaey, 2016-04-09:
In [this](https://github.com/matthijskooijman/arduino-lmic/blob/a5c12967d7e1b0f591438bad5557017bd135a4aa/src/lmic/radio.c#L224-L229) block of code, DIO mapping for LoRa mode is defined. In particular MAP_DIO2_LORA_NOP (which should represent ----11--, or 0x0C) is defined as 0xC0, which equals 11------. 

This sets the DIO0 mapping to 11 instead, corresponding to it being unused according to the [SX1272 datasheet](http://www.semtech.com/images/datasheet/sx1272.pdf) in table 17. This value is written into the DIO mapping register in [this](https://github.com/matthijskooijman/arduino-lmic/blob/a5c12967d7e1b0f591438bad5557017bd135a4aa/src/lmic/radio.c#L560) OR statement. As a result, when receiving in LoRa mode, a [check on DIO0](https://github.com/matthijskooijman/arduino-lmic/blob/a5c12967d7e1b0f591438bad5557017bd135a4aa/src/hal/hal.cpp#L68) does not signal when a receiving event has occurred.

I have not yet had the chance to test the effect of this change, when I do and it fixes any issues I had with receiving, I will make a pull request. In the mean time, I am posting this here to hear other people's opinions (I might be overlooking something, eg. other modules than SX1272).

I am aware this would be an error at the IBM side, but I'm making a GH issue here first, as this HAL makes use of the pin.

matthijskooijman, 2016-04-12:
I haven't looked closely at the code and datasheet yet, but it sounds like you did. I did however test this code on both SX1272 and SX1276 hardware, so it shouldn't be completely broken on either one, I believe.

matthijskooijman, 2016-04-16:
I had a closer look at things, and your analysis seems completely accurate to me - that particular define really seems incorrect. Knowing this, I can't really explain why RX or TX  is working at all. I could imagine that the SX chip takes a shortcut and implements the 11 DIO mapping the same as either 00 or 01 (rxdone or txdone), but that would only explain why either of them works, not both. Perhaps there is some timeout event that triggers a bit after txdone/rxdone would/should have triggered, that also checks the SX registers and finds the pending event and processes it? I don't think I've seen this in the code, but I haven't fully understood all of it.

Perhaps 11 for DIO0 is implemented as "rxDone when in RX mode, txDone when in TX mode"?

I guess the next step to investigate this would be to attach a logic analyzer to the DIO pins and see what actually happens with them?

msamaey, 2016-04-17:
My current assumption is that DIO0 in fact only signals a 'done' event, and the mapping is purely to keep it in line with the other pins:

- The decision to check the interrupt flags [hal_io_check()](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/hal/hal.cpp#L60-L72) is based on any change in the pin state, not specifically if DIO0 becomes high.
- In [radio_irq_handler()](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/radio.c#L734-L792) the interrupt flags are used to check if either RX or TX is done. CAD could also be checked by it's interrupt flag. 

My presumption is thus that even if one would map DIO0 to TXDone (01) mode whilst receiving, it would still signal correctly when a packet is received. This would only cause issues in libraries that do not check the itnerrupt flags but take action based on the pin state.

Using a logic analyzer would shed more light on the situation, but as the receiving of LoRa packets is only a part of our university project, I do not think we will be able to investigate it further due to tight deadlines.