matthijskooijman/arduino-lmic #16: EV_TX_COMPLETE slow to occur

torntrousers, 2016-04-23:
I've been looking at a battery powered node using an ESP8266 and RFM95W hoping to get quite long battery life sleeping most of the time, however it turns out that it takes quite a long time for the node to do a send and get the TX_COMPLETE event back, 6 - 10 seconds, so even when using deepSleep for the rest of the time that 10 seconds of up time would drain the batteries quite quickly.

That seems odd, it shouldn't take so long should it? Poking around in the LMIC code and the SX1276 datasheet it sounded like after a send DIO0 going high would indicate the send is complete so I've tried using that to trigger when to put the ESP8266 in deepSleep, and that seems to work, DIO0 goes high about 130 milliseconds after the send and putting the ESP8266 to sleep then the gateway still seems to get the transmission ok.

I wonder is this a bug in the LMIC code that its not doing the EV_TXCOMPLETE sooner, or perhaps is the SX1276 carrying on finishing the send after i put the ESP to sleep after 130milliseconds and the SX1276 stays awake? Any ideas?

I've put this code i have for this ESP8266/RFM95W node [here](https://github.com/HarringayMakerSpace/IoT/blob/master/LoraNode1/LoraNode1.ino).

(I've previously asked about this over [here](http://forum.thethingsnetwork.org/t/esp8266-rfm95w-node-how-long-should-a-send-take/1890))

matthijskooijman, 2016-04-25:
One thing that springs to mind is that TXCOMPLETE only triggers after the RX windows are also completed. However, that shouldn't take *that* long.

This might be a result of #13 perhaps - there is a (as of yet unfixed) bug in the DIO mapping, which doesn't seem to affect the transmission at least for me (though thinking about it again, I actually think that TX_COMPLETE might have taken 6-10 seconds in my testing as well, I just might never have realized that was a problem). Perhaps you could try the fix suggested in that issue (change `MAP_DIO2_LORA_NOP` to `0x0C`) and see if that helps?

torntrousers, 2016-04-25:
Thanks for the suggestion. I've tried that but it doesn't seem to help though and TXCOMPLETE still takes six seconds or so to happen.

tftelkamp, 2016-04-25:
TXCOMPLETE is purposely delayed, see lmic.c line 1498. It can be tuned using DNW2_SAFETY_ZONE.

torntrousers, 2016-04-25:
@tftelkamp mmm, so as an experiment I changed that:
```
//                            (os_getTime() + DNW2_SAFETY_ZONE + rndDelay(2)),
                            (os_getTime()),
```
And that gets TXCOMPLETE fired in a bit over 2 seconds, so better but still nothing near the 130 milliseconds with the DIO0 going high approach.

tftelkamp, 2016-04-25:
2 seconds seems correct to me, if there is no downlink message (then it could be shorter). The TXCOMPLETE event signifies the end of a complete TX/RX cycle, and has to wait for the RX2 slot to pass. I think the purpose is also to block a new TX within that period, to prevent collisions.

Btw, RXCOMPLETE seems a class B thing (only).

torntrousers, 2016-04-25:
But for a Class A node why does it need to wait at all?

If I just stick with the code i have using the DIO0 interrupt to get deepSleep much faster do you think it would cause any problems?

tftelkamp, 2016-04-25:
It’s part of Class A. The network operator needs to be able to send MAC commands. You can go into deepSleep between TX and RX1, and between RX1 and RX2.

Also see section 3.3.6 of the 1R0 specification.

torntrousers, 2016-04-25:
Ok so as long as the node doesn't send again too soon so as to violate section 3.3.6 then would then be any problem with the node going straight to sleep with the DIO0 interrupt is triggered? I don't have any code in the node to do anything with any received message so it seems a shame to stay a wake for it. Especially when that staying awake will massively reduce the batter life on the node - even with that mod to get it down to 3 seconds up time means batter life would be **23 times less** than with the DIO0 interrupt approach  (3 seconds / 130ms = 23)

matthijskooijman, 2016-04-25:
@torntrousers, I haven't double-checked this, but there is a hal_waitUntil function (or something similar). AFAIK whenever that is called, LMIC will be idle until the given time, and the MCU can go to sleep (provided it wakes up on a DIO interrupt). Perhaps you could see if that would help your usecase?

torntrousers, 2016-04-27:
@matthijskooijman, thanks, had a look at hal_waitUntil but can't see how to use that to do what i need.

I've read the 1R0 spec a bit more now. I think I appreciate it might be useful to have the receive window after the node transmits, but its not needed for this type of battery powered node, it kind of seems like there should be another class of node, or a configurable header option on the send to say if the node will do a receive.

So, TBH I like how fast the node goes back to sleep with this current approach i have triggered off DIO1 going high, and i can't see how that will cause any trouble or how anything can even know that the node is not listening for the receive window, so think I'll just stick with this code for now.

Oliv4945, 2016-05-25:
@torntrousers, if you want to connect to an operator you must be able to receive downstream messages because they will send you some config data (ex: frequency & SF to be used).
I think that the best thing to do is being interrupt driven by DIO as mentioned by @matthijskooijman.
I plan to modify the code for that but did not have time yet.

matthijskooijman, 2016-05-25:
@Oliv4945, if you do, be sure to share your code :-)

Oliv4945, 2016-05-25:
@torntrousers. I just checked on an ATMega328 and as soon as the RX2 window expires or the message is received the EV_TX_COMPLETE is triggered without any delay
@matthijskooijman I will, but not time for now, I have two PR to clean and submit before:
LMIC send weird ACKDown and I modified the code to take into account MAC commands in frame header or in payload.

torntrousers, 2016-06-07:
Hi @Oliv4945, interesting comment about must receiving downstream messages about things like the frequency and SF to be used. One attribute of the sleeping ESP8266 node code is that the ESP sleep/wake up is really a complete restart, no state is persisted, and it always starts and sends from the MCU initial state, so for example will always be using channel 0. So like that there really isn't any point staying awake to listen for a downstream message. 

(What do the specs say - is that sort of node operation violating something in the spec?)

matthijskooijman, 2016-06-16:
@torntrousers, the problem with not saving state is not so much the channel, but your session is restarted. When using OTAA, you need to rejoin after every sleep to set up a new session with new encryption keys, which seems like significant overhead. If using ABP, you will have a statically configured session, but your frame counters will be reset, making the server-side detect a replay attack and refuse your packets (unless you disable replay protection, which I think TTN supports now).

Looking back at the original topic of this issue - it seems that TX_COMPLETE is indeed 3 seconds (DNW2_SAFETY_ZONE) later than it could be, which is a bit troublesome for sleeping. Looking at the code, I think there is a simple fix for this: Instead of delaying the TX_COMPLETE event by DNW2_SAFETY_ZONE, just run the TX_COMPLETE directly (i.e. call `processDnData()` directly), and then use `txDelay()` to make sure not TX happens within the DNW2_SAFETY_ZONE. This allows the main program to schedule a new TX right away, but it won't be transmitted until the safety zone is passed.

torntrousers, 2016-06-17:
Right, I did [ask about](http://forum.thethingsnetwork.org/t/new-backend-how-to-connect/1983/94) how to get at the frame counters so they could be persisted, and also found [this post](http://forum.thethingsnetwork.org/t/arduino-lmic-library-updated/1295/10) about them. I might have a go at just using LMIC.seqnoUp/Dn directly to try to persist them over sleep/wakeup in the ESP's RTC memory.

I couldn't find where to make the change - _just run the TX_COMPLETE directly (i.e. call processDnData() directly)_ - if you could help out with that I can give it a try?

matthijskooijman, 2016-06-17:
> Right, I did ask about how to get at the frame counters so they could be persisted, and also found this post about them. I might have a go at just using LMIC.seqnoUp/Dn directly to try to persist them over sleep/wakeup in the ESP's RTC memory.

Yeah, just accessing those fields directly should work, I believe.

> I couldn't find where to make the change - just run the TX_COMPLETE directly (i.e. call processDnData() directly) - if you could help out with that I can give it a try?

That would be here: https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/lmic.c#L1500-L1503

Replace the marked lines (including the return) with `txDelay(os_getTime() + DNW2_SAFETY_ZONE, 2);` should be all that's needed.

torntrousers, 2016-06-17:
Yes accessing the seqnoUp field directly seems to work ok, I've updated that [node code](https://github.com/HarringayMakerSpace/IoT/blob/master/LoraNode1/LoraNode1.ino) to do that persisting the values in the ESP flash.
Haven't got to trying the processDnData() change yet though.

torntrousers, 2016-06-19:
I've tried that change around processDnData. That makes the wakeup-transmit-gotoSleep take 2400 milliseconds. That compares to less than 200 milliseconds with the going to sleep after DIO0 high approach, so still the difference of a battery powered node lasting months or years.

matthijskooijman, 2016-06-21:
Right, my suggestion was indeed to make TXCOMPLETE happen ASAP, while still waiting for the RX window to complete. If you do not want that, you'll need to make other changes.

rmoscetti, 2017-06-20:
I have a similar problem here.
I am developing a weather station based on an Adafruit feather 32u4.
Currently, I am testing my sketch without deepsleep mode.
My sketch collects data every 10 min and the payload is sent after 6 acquisitions (i.e. after 1 hour; at 0, 10, 20, 30, 40 and 50 min). Using this setting, TXCOMPLETE takes more than 5 minutes to occur. What is very strange is that if change the cycle setting (e.g. data acquisition every 1 min and data send every 6 min) the TXCOMPLETE is really faster (it occurs just in few seconds).

Regarding the following:
> That would be here: https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/lmic.c#L1500-L1503
> 
> Replace the marked lines (including the return) with txDelay(os_getTime() + DNW2_SAFETY_ZONE, 2); should be all that's needed.

Do you mean these lines?
>e_.devaddr = LMIC.devaddr,
>e_.oldaddr = oldaddr,
>e_.nonce = LMIC.devNonce-1,
>e_.mic = mic,

I am not a programmer, but I am trying to do my best; so, please explain me in the simplest way you can.

Thank you in advance.

Best Regards,
Roberto

commit 6e098863 (2017-07-24): Emit EV_TXCOMPLETE earlier

Before, when no packet was received in RX2, there would be a 3-5 second
delay (for EU868) before calling `processDnData()` and emitting
EV_TXCOMPLETE. This was the `DNW2_SAFETY_ZONE` to prevent a subsequent
TX to collide with a downlink packet that was ongoing but not locked on
to.

With this change, `txDelay()` is used to still delay the subsequent TX,
without delaying the `EV_TXCOMPLETE` event, which allows battery-powered
devices that wait for this event before sleeping to get an extra 3-5
seconds of sleep for each transmitted packet, improving battery life.

Clo [... cut]
closed 2017-07-24 by matthijskooijman