matthijskooijman/arduino-lmic #18: lmic is not sending enough NewChannelAns ?

Oliv4945, 2016-05-25:
Hi,

I worked with an operator to validate the stack on their network and we might have spotted an issue I want to discuss with you before sending a PR : when the operator is sending several `NewChannelReq` in one downstream message, it expects several `NewChannelAns` in the following upstream message.

I am not sure of the interpretation of the LoraWAN spec 1.0 (page 28, L8), which is
> The end-device acknowledges the reception of a NewChannelReq by sending back a NewChannelAns command.

Currently, the stack only answers one `NewChannelAns` for each frame, regardless the number of `NewChannelReq`  inside.   
Does somebody else thinks that if the stack receives X `NewChannelReq` in one dowlink frame, it should send X `NewChannelAns` in the next uplink frame ? If so, I will do a PR on the code I wrote.

matthijskooijman, 2016-05-29:
Interesting question. I don't really know what the correct answer to this is. I've asked on the Things Network slack, I'll forward any replies here. Perhaps you could ask your question on [The Things Network forum](http://forum.thethingsnetwork.org), there's a lot of LoRaWAN experts reading along there :-)

avbentem, 2016-05-29:
For future reference: the question was posted on http://forum.thethingsnetwork.org/t/interpretation-of-newchannelans-specification/2287