matthijskooijman/arduino-lmic #21: Autoconvert EUIs to little-endian

matthijskooijman, 2016-06-16:
The LMIC library expects EUIs for OTAA to be in the little-endian format that is used in LoRa transmissions. Currently, the examples just specify the EUIs as byte arrays, requiring the programmer to manually "reverse" the bytes to get a little endian number. Ideally, you would just specify a uint64_t for the EUI and let the compiler/code figure things out.

Using a uint64_t means that the data will be stored in architecture-specific order, so handling it needs to be different between little and big endian systems. Unfortunately the libc standard endianness conversion functions aren't available on AVR, so we should probably add a specific one just for EUIs / uint64_t.

Ideally, the conversion would happen at compiletime through a macro, so at runtime the bytes can just be copied directly from PROGMEM using memcpy_P as now (manually copying and inverting seems to result in bigger code).

Oliv4945, 2016-06-16:
I think that it is a good idea 👍 
We might have a look [here](http://www.fourwalledcubicle.com/files/LUFA/Doc/151115/html/group___group___endian_conversion.html#gaa3d310efe7cd6126823b8721852d7e56), it is a code for AVR

zoutepopcorn, 2016-08-27:
I use the ttn web dashboard, 3 x click and you have your EUI in the format :D.