matthijskooijman/arduino-lmic #22: OTAA does not work on 8MHz ?

Oliv4945, 2016-06-16:
Does anyone tested OTAA on 8 MHz ATMega328 ?
It works well on my 16 MHz Arduino shield, but I do not receive any message on my 8 MHz boards during join process, whatever is the SF. I checked the timings which seems to be OK, and the gateway log shows that `joinAccept` is answered.

I am a bit lost because in APB the downlink works well on RX1 and RX2 :-(

matthijskooijman, 2016-06-16:
Yup, I've run on 8Mhz succesfully for a project last week. I have been using the internal oscillator, which is fairly inaccurate, but in the mjs branch there is a change that allows compensating for this (by enlarging the RX window). Perhaps you could try the same thing, just to confirm whether it's a timing problem or not? To do so, add something like this to the setup:

    // Let LMIC compensate for +/- 1% clock error
    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);

tftelkamp, 2016-06-16:
I had the same issue, and this solved it for me.

matthijskooijman, 2016-06-16:
@tftelkamp, "this" being the clock error thing? Good to hear my work is already useful for others too :-)

tftelkamp, 2016-06-16:
Yes indeed. Very useful!

Oliv4945, 2016-06-19:
Hi, clever solution !
Nevertheless I selected high accuracy oscillator so my problem does not come from here. In fact I think it was a power supply issue : I used my node powered only by the FTDI chip, 20 mA is too weak, it works now with a battery. What I do not understand is why downlink worked in ABP and not OTAA :/

So I close the issue, sorry to bother you !

closed 2016-06-19 by Oliv4945

matthijskooijman, 2016-06-21:
Weird indeed. Also weird that TX works but not RX, since TX needs even more power. Perhaps RX is more sensitive to voltage drops/fluctuations than TX. Or perhaps the insufficient power somehow influenced your clock accuracy? During OTAA, the RX window happens after 5 seconds, while regular downlinks happen after 1 second (so with OTAA, any clock accuracy will cause 5x more drift). Oh well.

Oliv4945, 2016-06-25:
Yes, it is weird. But since I add a battery I do not have troubles so I presume it was that...