matthijskooijman/arduino-lmic #24: Fix to avoid unable to compile mjs branch with AES

hallard, 2016-06-23:
@matthijskooijman 
I'm tried to compile new mjs branch (for ESP8266) but it does not work. I listed there the problems and the fix I've done

* 1st point

using `USE_ORIGINAL_AES` in `config.h` fire multiple undefined reference.
```
S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -g  -Os -nostdlib -Wl,--no-check-sections -u call_user_start -u _printf_float -u _scanf_float -Wl,-static "-LS:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266/tools/sdk/lib" "-LS:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266/tools/sdk/ld" "-LS:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp/ttn-abp.ino.elf" -Wl,--start-group "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\sketch\ttn-abp.ino.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\aes\lmic.c.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\aes\other.c.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\aes\ideetron\AES-128_V10.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\hal\hal.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\oslmic.c.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\radio.c.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\SPI\SPI.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\ESP8266WiFi\ESP8266WiFi.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\ESP8266WiFi\ESP8266WiFiAP.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\ESP8266WiFi\ESP8266WiFiMulti.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\ESP8266WiFi\ESP8266WiFiSTA.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\ESP8266WiFi\ESP8266WiFiScan.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\ESP8266WiFi\WiFiClient.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\ESP8266WiFi\WiFiClientSecure.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\ESP8266WiFi\WiFiServer.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\ESP8266WiFi\WiFiUdp.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp/arduino.ar" -lhal -lphy -lpp -lnet80211 -llwip_gcc -lwpa -lcrypto -lmain -lwps -laxtls -lsmartconfig -lmesh -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-LC:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp"
C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o: In function `convFreq':
S:\Skydrive\devt\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:85: undefined reference to `AESAUX'
C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o:(.text.aes_cipher$part$3+0x4): undefined reference to `AESAUX'
C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o:(.text.aes_cipher$part$3+0x8): undefined reference to `AESAUX'
C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o:(.text.aes_cipher$part$3+0xc): undefined reference to `AESKEY'
C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o:(.text.aes_cipher$part$3+0x10): undefined reference to `os_aes'
C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o: In function `aes_cipher':
S:\Skydrive\devt\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:85: undefined reference to `os_aes'
C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o: In function `rxschedInit':
S:\Skydrive\devt\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:85: undefined reference to `os_aes'
C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o: In function `decodeFrame':
S:\Skydrive\devt\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:85: undefined reference to `os_aes'
C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o: In function `engineUpdate':
S:\Skydrive\devt\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:85: undefined reference to `os_aes'
C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o:S:\Skydrive\devt\Arduino\libraries\arduino-lmic\src\lmic/lmic.c:85: more undefined references to `os_aes' follow

collect2.exe: error: ld returned 1 exit status

Using library arduino-lmic at version 1.5.0+arduino-1 in folder: S:\Skydrive\devt\Arduino\libraries\arduino-lmic 
Using library SPI at version 1.0 in folder: S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\libraries\SPI 
Using library ESP8266WiFi at version 1.0 in folder: S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi 
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

Fix is in the file `src/aes/limc.c`, change at the beginning, the test `#if defined` after the include for the compiler knows the `USE_ORIGINAL_AES` and also change the `#include` file name (set te relative) because it was not found so 
```arduino
#if defined(USE_ORIGINAL_AES)
#include "oslmic.h"
```
becomes
```arduino
#include "../lmic/oslmic.h"
#if defined(USE_ORIGINAL_AES)
```

* Second point 

Now using `USE_IDEETRON_AES` in `config.h` fire a linker error with double definition of `aes_encrypt` only on ESP8266 but this is because SDK already contain `aes_encrypt()` function in `libcrypto.a`. 

```
S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -g  -Os -nostdlib -Wl,--no-check-sections -u call_user_start -u _printf_float -u _scanf_float -Wl,-static "-LS:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266/tools/sdk/lib" "-LS:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266/tools/sdk/ld" "-LS:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy  -o "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp/ttn-abp.ino.elf" -Wl,--start-group "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\sketch\ttn-abp.ino.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\aes\lmic.c.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\aes\other.c.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\aes\ideetron\AES-128_V10.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\hal\hal.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\lmic.c.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\oslmic.c.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\arduino-lmic\lmic\radio.c.o" "C:\Users\CHARLE~1\AppData\Local\Temp\build20a4a2935c21ce58917d68130563504a.tmp\libraries\SPI\SPI.cpp.o" "C:\Users\CHARLE~1\AppData\Local\Temp\b [... cut ...]
Using library arduino-lmic at version 1.5.0+arduino-1 in folder: S:\Skydrive\devt\Arduino\libraries\arduino-lmic 
Using library SPI at version 1.0 in folder: S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\libraries\SPI 
Using library ESP8266WiFi at version 1.0 in folder: S:\Skydrive\devt\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266WiFi 
exit status 1
Error compiling for board WeMos D1 R2 & mini.
```

Workaround was (may be better solution) to rename all `aes_encrypt()` to `limc_aes_encrypt()` to avoid target error due to existing function.

matthijskooijman, 2016-06-23:
Thanks for testing and reporting! I'll integrate both points when I merge this stuff to master, hopefully somewhere next week.

hallard, 2016-06-24:
Excellent, I will do a fork then
FYI node is working with Wifi Enabled, webserver and websockets ;-)
But I also needed to change 2 typdef in limc stack because unfortunatly the ip stack of ESP8266 (lwip) use same type `u8_t` and `s8_t` (and of course not defined same type and dependings on thing used for project it can refuse to compile complaining type already defined, I've done quick and dirty change from 
```
typedef uint64_t           u8_t;
typedef int64_t            s8_t;
```
to 
```
typedef uint64_t           u8l_t;
typedef int64_t            s8l_t;
```

Not sure it's not the best (strange that u8 typedef'd as uint64 by the way) but it works, worth mentioning this point.

I will now investigate on #7 with only one DIO connected

matthijskooijman, 2016-06-24:
Annoying duplicate typedefs :-)

I guess the LMIC library should just use the standard `uint8_t` types etc, there is no point (that I can see) in defining its own types. However, this is an invasive change, which should be made upstream, I think, so I'm not sure I'll make that already. Seeing that u8_t is only used in one place right now, guess changing that one isn't much of a problem.

> (strange that u8 typedef'd as uint64 by the way) 

8 bytes == 64 bits. But I see lwip [uses u8_t for uint8_t](https://github.com/kadamski/esp-lwip/blob/12bff2fe9619b6f0342e41296d01b404cd53f943/espressif/include/arch/cc.h#L52) which obviously conflicts.

hallard, 2016-06-25:
@matthijskooijman 
Got a good news, I made a working fix for #7 so now it works even with no physical connection on DIO pins ;-)

I'm giving below the modification, if you don't have time, I'll do a PR but I need to wait your merge of mjs branch to master.

Created in `radio.c` a new function `radio_has_irq()`

```arduino
// called by hal to check if we got one IRQ
u1_t radio_has_irq (void) {
    u1_t flags ;
    if( (readReg(RegOpMode) & OPMODE_LORA) != 0) { // LORA modem
        flags = readReg(LORARegIrqFlags);
        if( flags & ( IRQ_LORA_TXDONE_MASK | IRQ_LORA_RXDONE_MASK | IRQ_LORA_RXTOUT_MASK ) ) 
            return 1;
    } else { // FSK modem
        flags = readReg(FSKRegIrqFlags2);
        if ( flags & ( IRQ_FSK2_PACKETSENT_MASK | IRQ_FSK2_PAYLOADREADY_MASK) ) 
            return 1;
        flags = readReg(FSKRegIrqFlags1);
        if ( flags & IRQ_FSK1_TIMEOUT_MASK ) 
            return 1;
    }
    return 0;
}
```

Then in `hal_io_check()` if one of DIO pin is `LMIC_UNUSED_PIN`, look for IRQ by software like as follow

```arduino
static void hal_io_check() {
    uint8_t i;
    for (i = 0; i < NUM_DIO; ++i) {
        if (lmic_pins.dio[i] == LMIC_UNUSED_PIN)
        {
            // Check IRQ flags in radio module
            if ( radio_has_irq() ) 
                radio_irq_handler(0);
            
            // We've soft checked IRQ reading Radio register no need to continue
            // Setting this will exit us from for loop
            i = NUM_DIO;
        } else {
            if (dio_states[i] != digitalRead(lmic_pins.dio[i])) {
                dio_states[i] = !dio_states[i];
                if (dio_states[i])
                    radio_irq_handler(i);
            }
        }
    }
}
```

And comment the DIO assert in `hal_io_init()`
```
    //ASSERT(lmic_pins.dio[0] != LMIC_UNUSED_PIN);
    //ASSERT(lmic_pins.dio[1] != LMIC_UNUSED_PIN || lmic_pins.dio[2] != LMIC_UNUSED_PIN);
```

matthijskooijman, 2016-08-09:
I incorporated your suggested changes in the git master branch. Could you verify if it now works on ESP like this and, if so, close this issue?

hallard, 2016-08-09:
@matthijskooijman 
Thanks a lot, this will avoid me a PR, I'll test this ASAP
Charles

hallard, 2016-08-09:
@matthijskooijman 
Just to be sure, did you also included fix for #7 ?

matthijskooijman, 2016-08-09:
Nope, only the ESP fixes. I think supporting running without DIO pins is a bit of a feature request rather than a fix, and would warrant its own issue or PR.

hallard, 2016-08-10:
@matthijskooijman 
Got it, took time doing some test, just compile fine on ESP8266

In the main way, I've just submitted this [PR #34](https://github.com/matthijskooijman/arduino-lmic/pull/34) to add no DIO connection if needed.

matthijskooijman, 2016-08-10:
Thanks for testing. I'll close this issue, we can track the DIO thing in the PR you created. Thanks!

closed 2016-08-10 by matthijskooijman

hallard, 2017-11-06:
@matthijskooijman 
You wrote (I admit some time ago) that the changes are incorporated in master branch but I don't see them.
Did I missed something?

matthijskooijman, 2017-11-06:
The changes needed to run without DIO pins are not merged yet, but tracked in #34. Does that answer your question? Or are there other changes you are missing?

hallard, 2017-11-06:
No, sorry about this, my fault, since talking this on both issues this one is fine and closed, no problem ;-)