matthijskooijman/arduino-lmic #30: Renaming C files to cpp in order to have access to Serial.print() ?

Oliv4945, 2016-07-03:
It is really helpful for debugging :-)

matthijskooijman, 2016-07-04:
You can use printf if you uncomment [this line](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/config.h#L26). I guess that would be sufficient?

Oliv4945, 2016-07-04:
> You can use printf if you uncomment this line. I guess that would be sufficient?

If I am not wrong, it isn't. I just tried another time to be sure, and two steps are required:

1. Add `#include <Arduino.h>`
2. Rename the file to *.cpp

If not done I have this error:
>In file included from [...]libraries\arduino-lmic\src\lmic\oslmic.h:20:0,
                 from [...]libraries\arduino-lmic\src\lmic\lmic.h:18,
                 from [...]libraries\arduino-lmic\src\lmic\lmic.c:13:
[...]libraries\arduino-lmic\src\lmic\lmic.c: In function 'processJoinAccept':
[...]libraries\arduino-lmic\src\lmic\config.h:26:24: error: 'Serial' undeclared (first use in this function)
 #define LMIC_PRINTF_TO Serial

But maybe I missed something ?

matthijskooijman, 2016-07-04:
I said you can use `printf()`, not `Serial.print()` :-)

e.g.:

    printf("this is a test with a number: %d\n", some_number);

See also http://www.cplusplus.com/reference/cstdio/printf/

Oliv4945, 2016-07-12:
Ah, shame on me, even written in the code...To much Arduino those times :/
Sorry I shouldn't have bothered you with such question ! Issue closed 👍

closed 2016-07-12 by Oliv4945