matthijskooijman/arduino-lmic #33: TX power issues

matthijskooijman, 2016-08-09:
Currently, the library allows setting the TX power using `LMIC_setDrTxpow()`. This sets the `LMIC.adrTxPow` field, but this field is never actually used anywhere.

The actual tx power to use is kept in `LMIC.txpow` and is currently defined per-band for 868 and hardcoded for 900 (see `updateTx()` for both).

However, the actual writing of the registers in `configPower()` seems inconsistent. The comments say that PA_BOOST is used for SX1272 and RFO for SX1276, but then for both the PaSelect bit (0x80) is set in RegPaConfig to select PA_BOOST. However, the corresponding value in RegPaDac, which should be 0x7 for PA_BOOST is set to 0x4 (for RFO), or left at the default of 0x4.

Then, the cropping of the txpow value for 1276 seems wrong, if txpow is 16, it effectively becomes 0 instead of 15 (due to the `& 0xf` masking). The lower limit is set to 2, but if PA_BOOST is indeed not used, this should be 0.

Finally, the txpower -> register setting calculation is wrong, assuming no PA_BOOST is used. The datasheet says:

> Pmax=10.8+0.6*MaxPower [dBm]
> Pout=Pmax-(15-OutputPower) if PaSelect = 0 (RFO pins)

If MaxPower == 7, then Pmax = 15 and Pout = OutputPower (just like with SX1272, which does not have MaxPower). The calculation seems to assume this, but it sets MaxPower to 0.

matthijskooijman, 2016-08-09:
One more thing: LMIC should really allow selecting between PA_BOOST or RFO outputs, since most hardware only has either of them connected.

pprasan, 2016-11-24:
Hi Matthijs,

Just wondering, have there been any updates/fixes to setting the Tx power correctly? We're currently conducting experiments at 915MHz on the SX1272, and want to be able to control the Tx power.

Regards,
Prajwal.

matthijskooijman, 2016-11-24:
@pprasan, no progress yet, but I think @tftelkamp was planning to have a look at this soon.

a-andreyev, 2017-03-21:
Hello @matthijskooijman! Thank you for the project!
What's the current state of the issue for the SX1276? I'm using RFM95W. Is default configuration uses maximum boost available by module?

matthijskooijman, 2017-03-23:
AFAIK no progress has been made on this issue so far, the code still needs a thorough cleanup.

Additionally jmarcelino reported on slack that the RegPaDac register number is defined as 0x5A but it should be 0x4D. However, since the only value written to that register is the same as the default value, and since 0x5A is an unused address, operation is unaffected.

javobarra, 2018-01-11:
Hi @matthijskooijman!  Any news about this? Is it still hardcoded or was someone able to change it?
Thank you very much for updating LMIC!

cyberman54, 2018-01-28:
When i was checking different power settings on my test node today, and comparing the LMiC code with SX1276 data sheet, i came to the same findings as stated in the beginning of this thread.

For me the actual code seems clearly wrong, since on SX1276 it always sets PA output to the PA_BOOST pin. This is forced by OR-ing the value that is to be written to RegPaConfig register with 0x80:

`writeReg(RegPaConfig, (u1_t)(0x80|(pw&0xf)))`

I guess this can't work on board hardware which does not wire the PA_BOOST Pin #27 of SX1276, only using RFO_LF / RFO_HF (Pin #28 / #22) instead?

How do we take care of this issue?
Is this issue solved in LMiC 1.6?

cyberman54, 2018-01-28:
I opened [issue](https://github.com/mcci-catena/ibm-lmic/issues/15) in the IBM's repo on this.