matthijskooijman/arduino-lmic #35: ttn_abp.ino sample code always use channel 0?

hallard, 2016-08-11:
Since the new release, it seems that LIMC does not change channel on each send, Before this new update when I was using ttn_apb.ino example, the send was occurring from channel 0 to 8, today it's always channel 0.

I compiled sample code changing my keys and DIO mapping, I'm on a SAMD21 Arduino Zero with RFM95 
Here are my logs, and all seems consistent, packet are sent, received by gateway and sent to TTN, no problem.

**Serial Monitor**
[image]

** nodejs monitoring TTN entries**
[image]

**TTN Dashboard**
[image]

Did I missed something?

gizmocuz, 2016-08-11:
I don't experience this issue with the OTAA
The below log is just showing a few parts of an MQTT-Spy export i programmed, and got in hand quickly:

2016-08-11T12:00:30 1: up , freq: 868.30	rssi: -88	lsnr: 9.5
2016-08-11T12:01:36 1: up , freq: 867.10	rssi: -85	lsnr: 10
2016-08-11T12:02:41 1: up , freq: 868.50	rssi: -86	lsnr: 8
2016-08-11T12:03:46 1: up , freq: 867.30	rssi: -85	lsnr: 9.2
2016-08-11T12:04:53 1: up , freq: 868.10	rssi: -88	lsnr: 9.2
2016-08-11T12:05:59 1: up , freq: 867.50	rssi: -84	lsnr: 7.5
2016-08-11T12:07:04 1: up , freq: 868.30	rssi: -87	lsnr: 9.8
2016-08-11T12:08:10 1: up , freq: 867.70	rssi: -85	lsnr: 9.8
2016-08-11T12:09:16 1: up , freq: 868.50	rssi: -85	lsnr: 7.5
2016-08-11T12:10:22 1: up , freq: 867.90	rssi: -85	lsnr: 10
2016-08-11T12:11:29 1: up , freq: 868.10	rssi: -85	lsnr: 9.5
2016-08-11T12:12:35 1: up , freq: 867.10	rssi: -85	lsnr: 9.8
2016-08-11T12:13:42 1: up , freq: 868.30	rssi: -85	lsnr: 10
2016-08-11T12:14:48 1: up , freq: 867.30	rssi: -84	lsnr: 11
2016-08-11T12:15:54 1: up , freq: 868.50	rssi: -85	lsnr: 8
2016-08-11T12:17:00 1: up , freq: 867.50	rssi: -85	lsnr: 7
2016-08-11T12:18:07 1: up , freq: 868.10	rssi: -85	lsnr: 9.2
2016-08-11T12:19:14 1: up , freq: 867.70	rssi: -89	lsnr: 10.5
2016-08-11T12:20:20 1: up , freq: 868.30	rssi: -88	lsnr: 10
2016-08-11T12:21:27 1: up , freq: 867.90	rssi: -85	lsnr: 9
2016-08-11T12:22:32 1: up , freq: 868.50	rssi: -93	lsnr: 7.5
2016-08-11T12:23:38 1: up , freq: 867.10	rssi: -88	lsnr: 9.8
2016-08-11T12:24:45 1: up , freq: 868.10	rssi: -93	lsnr: 9.5
2016-08-11T12:25:51 1: up , freq: 867.30	rssi: -85	lsnr: 10.5
2016-08-11T12:26:57 1: up , freq: 868.30	rssi: -87	lsnr: 8.8
2016-08-11T12:28:03 1: up , freq: 867.50	rssi: -85	lsnr: 6.8
2016-08-11T12:29:10 1: up , freq: 868.50	rssi: -86	lsnr: 8
2016-08-11T12:30:17 1: up , freq: 867.70	rssi: -85	lsnr: 10.5
2016-08-11T12:31:23 1: up , freq: 868.10	rssi: -84	lsnr: 9.2
2016-08-11T12:32:29 1: up , freq: 867.90	rssi: -85	lsnr: 10.2
2016-08-11T12:33:36 1: up , freq: 868.30	rssi: -85	lsnr: 9.8

As you can see, its changing the frequency

hallard, 2016-08-11:
@gizmocuz 
thanks for your response, yeah, correct with OTAA it's working fine my OTAA dev UI for this devce is 00 04 A3 00 B0 00 00 01.
What is so strange is that I can see frame on TTN dashboard from same device when it was with atb sketch, but this device that is not anymore connected because I flashed the otaa version into ?!?!

[image]

[image]

hallard, 2016-08-11:
Ok got it, problem solved sketch was running on another node at job (in single channel mode) and was going thru another gateway to myapp.

changed deviceid in sketch, now all is fine, only node at job 60A7AB04 is in single channel ;-)

[image]

closed 2016-08-11 by hallard