matthijskooijman/arduino-lmic #36: Arduino/Genduino 101

rodriguise, 2016-08-12:
working with the sx1276 based rfm95 chip I have no issue using the UNO but when working with the Arudino 101 I get the following error:

```
Starting
FAILURE 
/Users/rob/Documents/Arduino/libraries/arduino-lmic/src/lmic/radio.c:708
VAL: 9
```

It is mentioned in the readme that the 101 has been tested. Were there any changes necessary to make this work?

rodriguise, 2016-08-12:
I found this PR to be relevant:
https://github.com/matthijskooijman/arduino-lmic/pull/27

I dropped the spi to 1Mhz and things seem to be working well now.

```
//hal.cpp
static const SPISettings settings(10E6, MSBFIRST, SPI_MODE0);
```

```
static const SPISettings settings(10E5, MSBFIRST, SPI_MODE0);
```

matthijskooijman, 2016-08-22:
Thanks for reporting back. I'll close this issue in favor of that PR.

closed 2016-08-22 by matthijskooijman