matthijskooijman/arduino-lmic #37: Error on Linkit ONE

idreamsi, 2016-08-19:
Hi
I using [Lora Shield](http://wiki.dragino.com/index.php?title=Lora_Shield) and [LinkIt ONE](http://wiki.seeedstudio.com/wiki/LinkIt_ONE), When I compile with the following error :
```
Arduino: 1.6.8 (Windows 7), Board: "LinkIt ONE"

Warning: platform.txt from core 'MediaTek ARM7 EJ-S (32-bits) Boards' contains deprecated recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{build.path}/{archive_file}" "{object_file}", automatically converted to recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} "{archive_file_path}" "{object_file}". Consider upgrading this core.
C:\Users\R@MiN\Documents\Arduino\libraries\arduino-lmic\src\hal\hal.cpp:77:14: error: 'SPISettings' does not name a type

 static const SPISettings settings(10E6, MSBFIRST, SPI_MODE0);

              ^

C:\Users\R@MiN\Documents\Arduino\libraries\arduino-lmic\src\hal\hal.cpp: In function 'void hal_pin_nss(u1_t)':

C:\Users\R@MiN\Documents\Arduino\libraries\arduino-lmic\src\hal\hal.cpp:85:13: error: 'class SPIClass' has no member named 'beginTransaction'

         SPI.beginTransaction(settings);

             ^

C:\Users\R@MiN\Documents\Arduino\libraries\arduino-lmic\src\hal\hal.cpp:85:30: error: 'settings' was not declared in this scope

         SPI.beginTransaction(settings);

                              ^

C:\Users\R@MiN\Documents\Arduino\libraries\arduino-lmic\src\hal\hal.cpp:87:13: error: 'class SPIClass' has no member named 'endTransaction'

         SPI.endTransaction();

             ^

C:\Users\R@MiN\Documents\Arduino\libraries\arduino-lmic\src\hal\hal.cpp: In function 'u4_t hal_ticks()':

C:\Users\R@MiN\Documents\Arduino\libraries\arduino-lmic\src\hal\hal.cpp:154:115: error: 'static_assert' was not declared in this scope

     static_assert(US_PER_OSTICK_EXPONENT > 0 && US_PER_OSTICK_EXPONENT < 8, "Invalid US_PER_OSTICK_EXPONENT value");

                                                                                                                   ^

exit status 1
Error compiling for board LinkIt ONE.
```
What's the solution ? 
Note : I found this PR to be relevant (But it did not work.):
https://github.com/matthijskooijman/arduino-lmic/pull/27

matthijskooijman, 2016-08-19:
It looks like the arduino corefor this Linkit board does not implement the entire API that the normal Arduino core does. The "SPISettings" class was added somewhere one or two years ago, so perhaps the Linkit folks haven't picked that up yet? Perhaps you're using an older version of it?

idreamsi, 2016-08-20:
Thanks @matthijskooijman