matthijskooijman/arduino-lmic #38: AdaFruit LoRa Feather M0 - ARM related Issue?

jortb, 2016-08-25:
I'm trying to get the AdaFruit M0 LoRa feather working with LoRaWAN with OTAA. While compiling the example I get this error. It seems to be ARM Cortex M0 related, but I haven't found a solution yet. Any ideas?

[ ARDUINO ERROR LOG]
Build options changed, rebuilding all
C:\Users\Jort-Signa\Documents\Arduino\libraries\IBM_LMIC_framework\src\hal\hal.cpp: In function 'void hal_printf_init()':

C:\Users\Jort-Signa\Documents\Arduino\libraries\IBM_LMIC_framework\src\hal\hal.cpp:223:54: error: '_FDEV_SETUP_WRITE' was not declared in this scope

     fdev_setup_stream (&uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE);

                                                      ^

C:\Users\Jort-Signa\Documents\Arduino\libraries\IBM_LMIC_framework\src\hal\hal.cpp:223:71: error: 'fdev_setup_stream' was not declared in this scope

     fdev_setup_stream (&uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE);

                                                                       ^

exit status 1
Error compiling for board Adafruit Feather M0 (Native USB Port).
[/ARDUINO ERROR LOG]

matthijskooijman, 2016-08-25:
Ah, completely forgot about this one. The printf implementation is AVR-specific right now. Could you try the https://github.com/matthijskooijman/arduino-lmic/tree/non-avr-printf branch (or better, cherry-pick the top commit of that branch onto master and try that)?

sillevl, 2016-08-26:
I'm not having this issue. I'm using the master branch of arduino-lmic, and version 1.6.11 of the Arduino IDE

mimiflynn, 2016-11-23:
I was getting this error and ended up deleting the LMIC library from my machine and reinstalling the version demarked by `-1` from the Arduino IDE Library Manager. I had to click the dropdown to find it. My code is uploading to the Feather M0 with no issue now.

jcwren, 2017-04-04:
I did have this issue, and pulled in the changes from the non-avr-printf branch. At first it appeared it wasn't working, but it turns out that stdout is in buffered mode, and the buffer is quite large.  I modified the code slightly to disable buffering and to make use of the ignored parameter in uart_putchar(). This also eliminates the need for calls to fflush() in user code.

I also added including stdio.h in src/lmic/radio.cpp and src/lmic/lmic.cpp to fix the warnings when compiling with PlatformIO.

```
static ssize_t uart_putchar (void *uart, const char *buf, size_t len)
{
  return ((Uart *) uart)->write (buf, len);
}

static cookie_io_functions_t functions =
{
  .read = NULL,
  .write = uart_putchar,
  .seek = NULL,
  .close = NULL
};

void hal_printf_init()
{
    if ((stdout = fopencookie (&LMIC_PRINTF_TO, "w", functions)))
      setvbuf (stdout, NULL, _IONBF, 0);
}
```

Cinezaster, 2017-06-27:
I copied the changes from the non-avr-printf branch. When compiling for an esp8266 nodemcuv2 I get these error messages when I change. Google doesn't give me good answers yet. 
It might help when you want to merge this branch, to know it doesn't compile for a esp8266
```
.pioenvs/nodemcuv2/lib/libarduino-lmic.a(hal.o):(.text._Z15hal_printf_initv+0x0): undefined reference to `_impure_ptr'
.pioenvs/nodemcuv2/lib/libarduino-lmic.a(hal.o):(.text._Z15hal_printf_initv+0xc): undefined reference to `fopencookie'
.pioenvs/nodemcuv2/lib/libarduino-lmic.a(hal.o): In function `hal_printf_init()':
hal.cpp:(.text._Z15hal_printf_initv+0x2c): undefined reference to `fopencookie'
collect2: error: ld returned 1 exit status
*** [.pioenvs/nodemcuv2/firmware.elf] Error 1
```
It is already been referenced here https://github.com/esp8266/Arduino/issues/925#issuecomment-232399678

gpickney, 2018-05-03:
Overwriting code in hal.cpp under `#if defined(LMIC_PRINTF_TO)` with the code that @jcwren provided allows me to use printf without the avr branch.

debaai, 2019-10-31:
I managed to fix this issue thanks to Matthijs Kooijman, by adding the latest commit of the non-avr-printf branch, with the minor modification that char needs to be converted to unsigned char when calling the LMIC_PRINTF_TO

`static ssize_t uart_putchar (void *, const char *buf, size_t len) {
    return LMIC_PRINTF_TO.write((const unsigned char*)buf, len);`