matthijskooijman/arduino-lmic #39: EV_JOINED doesn't fire.. little help?

JeffJam, 2016-08-26:
So... I have this library running on an OTAA sketch. It's running on an Uno R3. It's configured for CFG_us915.

Everything compiles and works nicely... or so it seems. From the network server I can see that the unit does a join request and that it's accepted, but the EV_JOINED event never fires.

I put in some print statements to give an idea of what is going on. Any input would be greatly appreciated.

```
Setup printf
Init...
RXMODE_RSSI
Reset...
Start send and join...
238: engineUpdate, opmode=0x8
Packet queued
Event
790: EV_JOINING
971: engineUpdate, opmode=0xc
Get ARTEUI 
Get DEVEUi 
Get DEVKEY 
2017: TXMODE, freq=902300000, len=23, SF=7, BW=125, CR=4/5, IH=0
315897: RXMODE_SINGLE, freq=923300000, SF=7, BW=500, CR=4/5, IH=0
Get DEVKEY 
Get DEVKEY 
377557: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
387245: engineUpdate, opmode=0xc
405393: engineUpdate, opmode=0xc
Get ARTEUI 
Get DEVEUi 
Get DEVKEY 
405786: TXMODE, freq=903000000, len=23, SF=8, BW=500, CR=4/5, IH=0
717579: RXMODE_SINGLE, freq=923300000, SF=7, BW=500, CR=4/5, IH=0
Get DEVKEY 
Get DEVKEY 
779238: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
788927: engineUpdate, opmode=0xc
829634: engineUpdate, opmode=0xc
Get ARTEUI 
Get DEVEUi 
Get DEVKEY 
830027: TXMODE, freq=902300000, len=23, SF=8, BW=125, CR=4/5, IH=0
Setup printf
Init...
RXMODE_RSSI
Reset...
Start send and join...
238: engineUpdate, opmode=0x8
Packet queued
Event
790: EV_JOINING
971: engineUpdate, opmode=0xc
Get ARTEUI 
Get DEVEUi 
Get DEVKEY 
2017: TXMODE, freq=902300000, len=23, SF=7, BW=125, CR=4/5, IH=0
315897: RXMODE_SINGLE, freq=923300000, SF=7, BW=500, CR=4/5, IH=0
Get DEVKEY 
Get DEVKEY 
377557: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
387245: engineUpdate, opmode=0xc
405393: engineUpdate, opmode=0xc
Get ARTEUI 
Get DEVEUi 
Get DEVKEY 
405786: TXMODE, freq=903000000, len=23, SF=8, BW=500, CR=4/5, IH=0
717579: RXMODE_SINGLE, freq=923300000, SF=7, BW=500, CR=4/5, IH=0
Get DEVKEY 
Get DEVKEY 
779238: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
788927: engineUpdate, opmode=0xc
829634: engineUpdate, opmode=0xc
Get ARTEUI 
Get DEVEUi 
Get DEVKEY 
830027: TXMODE, freq=902300000, len=23, SF=8, BW=125, CR=4/5, IH=0

```

gizmocuz, 2016-08-26:
in the setup function after LMIC_reset call, could you issue

`LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);`

That worked for me

JeffJam, 2016-08-26:
@gizmocuz thanks so much for your reply. I added the code you suggested but I'm getting similar results:

```
Setup printf
Init...
RXMODE_RSSI
Reset...
Start send and join...
235: engineUpdate, opmode=0x8
Packet queued
Event
786: EV_JOINING
967: engineUpdate, opmode=0xc
Get ARTEUI 
Get DEVEUi 
Get DEVKEY 
2013: TXMODE, freq=902300000, len=23, SF=7, BW=125, CR=4/5, IH=0
315893: RXMODE_SINGLE, freq=923300000, SF=7, BW=500, CR=4/5, IH=0
Get DEVKEY 
Get DEVKEY 
377553: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
387241: engineUpdate, opmode=0xc
424121: engineUpdate, opmode=0xc
Get ARTEUI 
Get DEVEUi 
Get DEVKEY 
424450: TXMODE, freq=903000000, len=23, SF=8, BW=500, CR=4/5, IH=0
736244: RXMODE_SINGLE, freq=923300000, SF=7, BW=500, CR=4/5, IH=0
Get DEVKEY 
Get DEVKEY 
797904: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
807593: engineUpdate, opmode=0xc
883313: engineUpdate, opmode=0xc
Get ARTEUI 
Get DEVEUi 
Get DEVKEY 
883706: TXMODE, freq=907300000, len=23, SF=8, BW=125, CR=4/5, IH=0
1199640: RXMODE_SINGLE, freq=923900000, SF=8, BW=500, CR=4/5, IH=0
1262396: RXMODE_SINGLE, freq=923300000, SF=12, BW=500, CR=4/5, IH=0
1272023: engineUpdate, opmode=0xc
1292578: engineUpdate, opmode=0xc
Get ARTEUI 
Get DEVEUi 
Get DEVKEY 
1292982: TXMODE, freq=904600000, len=23, SF=8, BW=500, CR=4/5, IH=0
```

hallard, 2016-08-27:
My 2 cents, when JOINED does not fire 
- be sure that there is a working GW in the area
- double and triple check keys devEUI, appEUI and appSession keys (if OTAA), take care of order for first 2 one, LSB first, so reversed order than TTN dashboard display

Each time it was not working on my side, it was a mistake on one of them!

I modified sketch to display all the keys at startup, really help to find out such kind of problems

JeffJam, 2016-09-02:
@hallard 
Thanks so much for the reply.

I actually own the gateway and it's within ~20'. 
I have verified that the devEUI etc are correct; when I change them to something that is known to be incorrect I can see there is an error in the gateway logs

MichelDeVittori, 2016-12-26:
@JeffJam resolved ?

Thanks, MDV