matthijskooijman/arduino-lmic #40: 1272 uno and error radio.c:659

cyryllo, 2016-09-01:
I use module sx1272 and i get failure 
```
I get error
Starting
FAILURE 
/home/cyryl/Arduino/libraries/arduino-lmic-1.5.0-arduino-1/src/lmic/radio.c:659
```

matthijskooijman, 2016-09-01:
Did you set the radio type in config.h to 1272? What version are you using exactly, since git master does not have an assert on that line...

ubergesundheit, 2016-09-01:
I am seeing the same error. It definitely worked last week. Since then I had some updates to the Arduino IDE Software.

I am using Arduino 1.6.11 from AUR on Arch.

Maybe this could be related?

gizmocuz, 2016-09-02:
This issue could be related to the recently updated AVR board definitions.

If you are running AVR board defs > 1.6.11, downgrade to 1.6.11

Try downgrading this  (tools->board->board manager, Arduino AVC Board, install 1.6.11)

There have been no code changes since more then 3 weeks, so i suspect the above

ubergesundheit, 2016-09-02:
I tried with 1.6.11. Same error.

JeffJam, 2016-09-02:
I have seen this error before! It happened to me (also with a UNO) when my CS pin came loose. You might also want to check your lmic_pinmap to make sure you're specifying the correct pins.

HTH

matthijskooijman, 2016-09-02:
> I have seen this error before!

Note that the error is just a failed assertion, so it can mean various things depending on the filename and line number shown. Unless anyone seeing this error actually looks up the line mentioned in *their version* of the source (or at least says what version they are using), I can't really start to provide any advice on fixing this.

hallard, 2016-09-02:
@matthijskooijman is right, unless you show here what's on line 689 of `radio.c`  and your LMIC pins definition of your sketch we're absolutely blind and we can't give any advice...

Did you changed the LMIC pins definition in your sketch to conform to Dragino shield wiring?

ubergesundheit, 2016-09-02:
I use the latest version(90bc0492b632475ae918fc9c65cd5508f1fdb1af) from this git repo. Currently the line is https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/radio.c#L689

I am somewhat sure this is related to an update with either the arduino ide or the avr boards as the same sketch worked last week. I also tried another board. It has the same issue

matthijskooijman, 2016-09-03:
@ubergesundheit, it is possible your error is different from the one of the original reporter. In your case, the hardware is not detected as a 1276, which likely means that the SPI communication is failing. Regarding the Arduino update, 1.6.10 upgraded the compiler version, so it might very well be caused by that. I don't have any hardware handy to test now, though.

matthijskooijman, 2016-09-03:
@ubergesundheit, it might be good to download an older copy of the Arduino IDE (1.6.9 for example) and see if it does work with that one now. If you download the zip version, there is no need to install anything, just run them side-by-side.

cyberman54, 2016-09-04:
I got the same Error Message with identical Line number in Radio.c when trying to join by OTAA, using an Ideetron Nexus Board with ATmega328p. ABP with the same Code works. So the Error probably has some context with OTAA joining.

hallard, 2016-09-04:
@cyberman54 
The error is when the stack is talking to RFM95 (or other) with SPI.
The error is because the version of module read from device does not match with the one it should receive, for SX1276 (RFM95) you should receive be `0x12` and for SX1272 it should be `0x22`. It would be interesting to see what value you have, for this change in `radio.c` by adding a

```arduino
#ifdef CFG_sx1276_radio
    ASSERT(v == 0x12 );
#elif CFG_sx1272_radio
    ASSERT(v == 0x22);
#else
#error Missing CFG_sx1272_radio/CFG_sx1276_radio
#endif
```

to 

```arduino
    Serial.print("Read chip version=0x");
    Serial.println(v,HEX");
#ifdef CFG_sx1276_radio
    ASSERT(v == 0x12 );
#elif CFG_sx1272_radio
    ASSERT(v == 0x22);
#else
#error Missing CFG_sx1272_radio/CFG_sx1276_radio
#endif
```

That said the problem is one of :
 
- Arduino IDE changed things on SPI between version and SPI don't works anymore (I clearly don't think so)
- Wiring to module is wrong (CLK/MOSI/MISO and also CS)
- Wiring definition `LMIC_pins`  is you example (the sketch) as not been adapted to the board you're using

Another issue could be that SPI is going to fast to your device, try switching this [line](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/hal/hal.cpp#L77) in `hal.cpp` with 
```arduino
static const SPISettings settings(4E6, MSBFIRST, SPI_MODE0);
```

hallard, 2016-09-04:
Looking at the schematics For [Ideetron Nexus Board](http://webshop.ideetron.nl/Files/3/1000/1211/Attachments/Product/4778I8m8OV005G7UF6f3815000Fu1Z48.pdf) wiring should be in example sketch :

```arduino
// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LMIC_UNUSED_PIN,
    .dio = {4, 5, 7}
};
```

hallard, 2016-09-04:
Looking at the schematics For [Dragino Lora Shield V1.2+]https://github.com/dragino/Lora/blob/master/Lora%20Shield/hardware/v1.3/Lora%20Shield%20v1.3.sch.pdf) wiring should be in [example sketch](https://github.com/matthijskooijman/arduino-lmic/blob/master/examples/ttn-otaa/ttn-otaa.ino#L61-L67) :

```arduino
// Pin mapping Dragino Shield V1.2 and V1.3
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LMIC_UNUSED_PIN,
    .dio = {2, 6, 7}
};
```

If using Dragino Shield Version V1.1 DIO1 and DIO2 are not connected you need to 

- connect DIO1 and DIO2 with Wiring to D6 (DIO1) and D7 (DIO2) (code remains same as one above)
- or use the pool IRQ version [Pull Request](https://github.com/matthijskooijman/arduino-lmic/pull/34) and use the code below

```arduino
// Pin mapping Dragino Shield V1.1
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LMIC_UNUSED_PIN,
    .dio = {2, LMIC_UNUSED_PIN, LMIC_UNUSED_PIN}
};
```

cyberman54, 2016-09-04:
It came out, it was as simple pin mapping problem. I did not tailor the pin mappings in the example to the nexus board pin mapping properly. Sorry for issuing that here, my fault. Thank you for your support.

sergiuszm, 2016-11-17:
@hallard Hi Hallard, could you help me with LoRa Bee [schematic](http://wiki.dragino.com/images/a/a2/Lora_BEE.sch.png) which is placed on Parallax Xbee Adapter board [schematic](https://www.parallax.com/sites/default/files/downloads/32403-XBee-Adapter-Board-v1.1.pdf)?

sergiuszm, 2016-11-19:
I was wrong as I though. I have found a better schema in which I have connected:
- nss to pin 10
- DIO1, DIO2 and DIO 3 to pins: 2, 3, 4
- rst to pin 5

Unfortunately, I have the same problem still as in this thread.

matthijskooijman, 2016-11-19:
Since the original reporter has not provided any additional information, and the comments have possibly strayed very far from the original issue (I suspect we have been talking about two different failed assertions here), I'm closing this issue. For future reference, note that two assertion error messages might be completely different issues if they refer to different lines. Hence saying "I also have an assert failure" does not really mean anything - be sure to include the line number *and* the contents of that line in any issue you create.

closed 2016-11-19 by matthijskooijman

hendabenayed, 2017-03-14:
I use module sx1276 on uno and i get failure

Starting
FAILURE
\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:689
any help please

cyberman54, 2017-03-14:
Check your pinout and pin definitions.

hendabenayed, 2017-03-15:
i use lora gps hat for arduino uno and i get the same error
Starting
FAILURE
\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:689
 are this module incompatible with arduino uno??

Roberto6969, 2017-03-15:
I use http://www.dragino.com/products/lora/item/109-lora-bee.html with AT328 CPU (not UNO board, only CPU with minimum components).
When i send message for the first time it goes through, but in the second attempt get error FAILURE \Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:660.
Resources of the AT328 are on the limit but i believe this is not the problem because Dragino sends first message.
Any idea how to send more messages than single one?

cattvm, 2017-03-15:
The error is because of the pin on the IC is not configured correctly. Can
you share your code for Pin configuration a in the method of programming
your using? either is it Arduino or Atmel Studio?!

Regards,

Ashish M Mahendra
Ph No: +91 99868 85788.
Skype: ashi.mahe@hotmail.com

cyberman54, 2017-03-15:
again: Check your DIO Pin mapping and ensure that pins needed for lmic are properly connected on your shield to RFM95 module.

Roberto6969, 2017-03-15:
@cattvm Do you mean this part:
// Pin mapping
const lmic_pinmap lmic_pins = {
  .nss = 10,  // 10 Connected to pin D10
  .rxtx = LMIC_UNUSED_PIN,  // 0 
  .rst = 9,   // 9 Needed on RFM92/RFM95? (probably not)
  .dio = {2, 6, 7}, // 2,6,7Specify pin numbers for DIO0, 1, 2
  // connected to D4, D5, D7 on Arduino
};

@cyberman54 What does it mean "properly" connected? Is there any particular advice? Should be - for example - reset pin connected or not?

cyberman54, 2017-03-15:
yes, this is the relevant part. The pin mappings in table must match pin mappings of used hardware. You should check if your LoRa shield has appropriate pin mapping and if it features all DIO pins which are needed by lmic. Consult IBM lmic 1.5 documentation pdf to find out which pins are needed.

Note: there is no standard to map the RFM95 pins to I/O ports of your CPU. Nearly every shield with RFM95 uses different pin mappings, and some (e.g. Adaptive Dragino shield) do not connect all pins which are necessary for using lmic. In this case you either need to tinker your hardware, or use another LoRaWAN software stack.

cyberman54, 2017-03-15:
addendum: if you tinker your hardware to add pin mappings, be aware of different voltage levels of Arduino board (mostly 5V) and RFM95 LoRa shield (often 3,3V). If voltages do not match, you may want to use level shifter to protect your RFM95 from damage.

cyberman54, 2017-03-15:
extract from IBM lmic 1.5 documentation:

---
3.1 HAL Interface
The following groups of hardware components must be supported:
 Four digital I/O lines are needed in output mode to drive the radio’s antenna switch (RX and 
TX), the SPI chip select (NSS), and the reset line (RST).
 Three digital I/O lines are needed in input mode to sense the radio’s transmitter and receiver 
states (DIO0, DIO1 and DIO2).
 A SPI unit is needed to read and write the radio’s registers.
 A timer unit is needed to precisely record events and to schedule new protocol actions.
 An interrupt controller is needed to forward interrupts generated by the digital input lines.
---

=>  check, if your hardware fulfills these requirements
=>  check, if pin mappings in your code match pin mappings of your hardware

With correct pin mappings this code works pretty good and stable. IBM lmic is a full LoRaWAN compliant software stack.

cyberman54, 2017-03-15:
erratum: Adaptive LoRa shield is not compatible with lmic due to missing pin connection, do not know if Dragino shield works.

This one did not work out of the box, but is able to be tinkered with level shifter:

http://adaptivedesign.ie/products/rf95-868mhz-lora-shield/

Roberto6969, 2017-03-20:
@cattvm and @cyberman54 thank you for your help - it was really helpful! I am finally back to lab so i checked connections and SW configuration. To help anyone who will fall in such problems i will highlight roots of my problem.

Lorashield http://www.dragino.com/products/module/item/102-lora-shield.html works fine with following configuration:

// Pin mapping
const lmic_pinmap lmic_pins = {
  .nss = 10,  // 10 Connected to pin D10
  .rxtx = LMIC_UNUSED_PIN,  // 0 For placeholder only, Do not connected on RFM92/RFM95
  .rst = 9,   // 9 Needed on RFM92/RFM95? (probably not)
  .dio = {2, 6, 7}, // Specify pin numbers for DIO0, 1, 2

Lorashield is "shield" so you can not physically change pins (in this case this is good ;-))

LoraBee is other story particulary if you build copy of Arduino with minimum parts. In this case is up to you how you will connect pins. And here is important that pin mapping follows connected pins correctly. In my case the configuration is:

// Pin mapping
const lmic_pinmap lmic_pins = {
  .nss = 10,  // 10 Connected to pin D10
  .rxtx = LMIC_UNUSED_PIN,  // 0 For placeholder only, Do not connected on RFM92/RFM95
  .rst = 9,   // 9 Needed on RFM92/RFM95? (probably not)
  .dio = {4, 5, 7}, // Specify pin numbers for DIO0, 1, 2

Still open question is if RESET is needed or not. I have done one test only without connected RESET and i got FAILED after 164 frames sent. So i connected it and now works.

And as @cyberman54 highlight - take care about voltage level. LoraBee is not 5V tolerant! Voltage converter is good solution but working with ATMEL on 3.3V is even better particulary if you targeting low power.

yotha, 2017-04-25:
Hello, I use an arduino Uno with a RFM95W (LoRa module). When i send message for the first time the packet is queued, but I get an error FAILURE \Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:660.

// Pin mapping
const lmic_pinmap lmic_pins = {
.nss = 10, // 10 Connected to pin D10
.rxtx = LMIC_UNUSED_PIN, // 0
.rst = 7, // 7 Needed on RFM92/RFM95? (probably not)
.dio = {2, 3, 4}, // 2,3,4 Specify pin numbers for DIO0, 1, 2
// connected to D2, D3, D4 on Arduino
};

Any idea how to send a messages correctly ?

cyberman54, 2017-04-25:
What type/brand of LoRa Module are you using?
- Wiring?
- Levelshifting?

yotha, 2017-04-25:
I am using a RFM95W 868Mhz RF Transceiver Module. I communicate in SPI between the module and the Arduino Uno.

cyberman54, 2017-04-25:
SPI is not sufficient for using LMIC. You need additional interrupt/signalling wiring (at least D0, D1, D2). You to ensure that these connections between your RFM95 board and your Arduino are present, correctly annotated in LMIC code and that 3,3v/5v level shifting is installed.

yotha, 2017-04-25:
I am using D0,D1, D2, nss and rst on my wiring, this is my pin mapping 
// Pin mapping
const lmic_pinmap lmic_pins = {
.nss = 10, // 10 Connected to pin D10
.rxtx = LMIC_UNUSED_PIN, // 0
.rst = 7, // 7 Needed on RFM92/RFM95? (probably not)
.dio = {2, 3, 4}, // 2,3,4 Specify pin numbers for DIO0, 1, 2
// connected to D2, D3, D4 on Arduino
};
I have connected MISO, MOSI, SCK, GND and 3.3v of course.

cyberman54, 2017-04-25:
Looks like a pin mapping issue, double-check your wiring.
And ensure that correct chip SX1272 vs. SX1276 is selected, like the author explained before in this thread.

Roberto6969, 2017-04-25:
Use correct names of pins! On CPU ATmega328 is D5 (pin 11), D6 (pin 12) and D7 (pin 13). So in case you connect D0 of RFM95 to D5, D1 to D6 and D2 to D7 you should use: .dio = { 5, 6, 7}  // pin names and not numbers. With other words: physical pin on Atmel328P no. 11 is D5 on Arduino UNO and if you connect this D5 from UNO to D0 on RFM95 you should use .dio = { 5, ... and so on.

yotha, 2017-04-25:
Ok so I replace the pin mapping by the physical pin number  but now I got this error  Failure : /Arduino/libraries/arduino-lmic-master/src/lmic/radio.c:689
This is my new pin mapping =>
// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 16,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 13,
    .dio = {4, 5, 6},
};

trlafleur, 2017-04-25:
If I remember correctly, D0 needed to be connected to a pin that supports interrupt... i.e.: pin 2

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

Roberto6969, 2017-04-25:
Here is working config:
const lmic_pinmap lmic_pins = {
  .nss = 10,  // 10 Connected to pin D10
  .rxtx = LMIC_UNUSED_PIN,  // 0 For placeholder only, Do not connected on RFM92/RFM95
  .rst = 9,   // Needed on RFM92/RFM95??? 
  .dio = {5, 6, 7}, // Specify pin numbers for DIO0, 1, 2 - connected to D5, D6, D7 on Arduino
};
Do not use physical pin numbers - use AVR names https://i.stack.imgur.com/dVkQU.jpg

trlafleur, 2017-04-25:
If you look at the code at 659, you can see that it's trying to read a version register on the chip to make sure it's the correct radio.... it fails, so your not connected correctly to the radio yet...

#ifdef CFG_sx1276_radio
ASSERT( v == 0x12 );
#elif CFG_sx1272_radio
ASSERT(v == 0x22);

Here is my mapping I use with a different processor,  I'm using a Moteino Mega LoRa with radio attach to the board.  ( rst and d3 are not needed for LoRa if your using RFM95, but it's ok to connect them...) I stoped using 328 processor a few years back... too small, flash, ram ect, most everything now is SAMD21G18A, 32 bits, faster, more flash, ram, lower power, cheaper device..

Look at using a 32 bit RocketScream M0 pro or Adafruit Feather LoRa M0, or the MoteinoMega LoRa, you won't go back to using the 328 CPU 

Hope this helps...

#ifdef __AVR_ATmega1284P__      // MoteinoMega LoRa
// Pin mapping
lmic_pinmap pins =
{
.nss = 4,
.rxtx = 0, // Not connected on RFM92/RFM95
.rst = 0, // Needed on RFM92/RFM95 ??
.dio = {2, 22, 0},
};
#define myLED 15
#define MyFlashCS 23

#elif __SAMD21G18A__ // RocketScream
// Pin mapping
lmic_pinmap pins =
{
.nss = 5,
.rxtx = 0, // Not connected on RFM92/RFM95
.rst = 0, // Needed on RFM92/RFM95 ??
.dio = {2, 6, 0},
};
#define myLED 13 // Is 13 on M0
#define MyFlashCS 4
#else
#error Wrong Processor defined
#endif

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

Roberto6969, 2017-04-25:
@trlafleur Would you be so kind and post link to CPU or development board you use? 328P is a little bit short for use LMIC library with some other sensor libraries

trlafleur, 2017-04-25:
As I pointed out, I stopped using ATMega328 processor a few years back...
too small, flash, ram ect, most everything now is M0, SAMD21G18A, 32 bits,
faster, more flash, ram, lower power, cheaper device..

My favorite 32 bit processor board:
RocketScream M0 ultra pro Ver2, RFM69 or RFM95 radio, battery connector,
EUI64 chip, large external flash, great support...
http://www.rocketscream.com/blog/product/mini-ultra-pro-v2-with-radio/

Other 32 Bit:
Adafruit Feather LoRa M0, NO EUI64, No External flash, battery connector, RFM69
or RFM95
https://www.adafruit.com/product/3178
https://www.adafruit.com/product/3176

Non 32 bit processors:
MoteinoMega LoRa, ATmega1284P, RFM69 or RFM95, EUI64 chip, large external
flash
https://lowpowerlab.com/shop/product/119

Moteino LoRa, ATmega328P, RFM69 or RFM95, large external flash, NO EUI64
chip
https://lowpowerlab.com/shop/product/99

Adafruit Feather LoRa, ATmega32U4 CPU, NO EUI64, No External flash, battery
connector, RFM69 or RFM95
https://www.adafruit.com/product/3078

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

yotha, 2017-04-25:
All it's working now, I send a message and the gateway receive it.

Thx all of you, my problem was about of a bad welding..

cyberman54, 2017-04-25:
i told you: double-check you wirings...

VanXungUIT, 2017-07-09:
Hello!
Can I help you?

I configure and connected with board Arduino 
[image]
 And result when I running: 
[image]

It just stood there.
Is it waiting for the gateway's response?

pierrot10, 2017-11-03:
> I have seen this error before! It happened to me (also with a UNO) when my CS pin came loose. You might also want to check your lmic_pinmap to make sure you're specifying the correct pins.
> HTH

I confirmed, I solved the problem by correcting the lmic_pinmap

necessaryevil86, 2020-05-13:
Hello, I use an RFM95 module. I also get an error in the serial monitor: 
> Starting
> FAILURE 
> D:\Dropbox\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:689

**line 686 ... 694 in radio.c:**
> // some sanity checks, e.g., read version number
>     u1_t v = readReg(RegVersion);
> #ifdef CFG_sx1276_radio                     //<--- line 689
>     ASSERT(v == 0x12 );
> #elif CFG_sx1272_radio
>     ASSERT(v == 0x22);
> #else
> #error Missing CFG_sx1272_radio/CFG_sx1276_radio
> #endif
> 

**Hardware:**
- Arduino Pro Mini 3.3V (clone)
-RFM95 (RF-module)
-diycon PCB,  [https://diycon.nl/product/lora-node-pcb-203-interface-pro-mini-rfm92-95](url)
-Arduino pin 2, 8 and 9 connect to DIO0, DIO1 and DIO2 on the RFM95
-Arduino pin10 connects to .nss on the RFM95

**Software:**
Pin mapping:
// Pin mapping
const lmic_pinmap lmic_pins = {                                                                                                               
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LMIC_UNUSED_PIN
    ,
    .dio = {2, 8, 9},
};

Config.h line 8... 15

> #define CFG_eu868 1
> //#define CFG_us915 1
> // This is the SX1272/SX1273 radio, which is also used on the HopeRF
> // RFM92 boards.
> //#define CFG_sx1272_radio 1
> // This is the SX1276/SX1277/SX1278/SX1279 radio, which is also used on
> // the HopeRF RFM95 boards.
> #define CFG_sx1276_radio 1

**Other**
I tried Arduino IDE 1.8.12 and Arduino IDE 1.6.9 without any luck.

Any ideas?

pierrot10, 2020-05-14:
Hello necessaryevil86

I did as the following with an Adafruit Feather MO

```
CS/NSS => 5
RST => 6
IRQ (DIO_0) => 10
DIO_1 => 11
DIO_2 => 12

// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 5,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 6,
    .dio = {10, 11, 12},
};
```
That works since a while. 
I things, I tried with an Pro Mini (I do not remember) as well, 2 years ago, but I used the above mapping.

I hope it can help.

necessaryevil86, 2020-05-14:
Hello pierrot10,
Thank you for your response. I send my sketch to a friend, and he got it working. So it seems to be a hardware problem. Also, I'm using a pcb, so I'm tied to the corresponding pin mapping.

cristianomazzuoli, 2020-06-20:
Hi all, 
I had the same error but only on one of two "almost" identical boards, except for the levels adaptation from 5V to 3V3.
I'am using two boards equipped with arduino nano (operating at 5V) ad a RFM96 (SX1276) radio module (operating at 3v3) each, so I have to lower the voltage of the arduino
outputs line from 5v to 3v3, as the RFM96 module isn't 5V tolerant.

The difference between the two board is that in the working one I use 4 level shifter from 5V to 3v3V for the lines MOSI, NSS, RESET, SCK, as the link below:
https://www.tme.eu/it/en/details/sn74lvc1t45dckt/interfaces-others-integrated-circuits/texas-instruments/

Instead, in the not working board I adapt this voltage using 4 simple resistor divider, with 1K in series from the arduino pin and 1k8 to ground, so I obtain (5V * 1K8) / 2K8 = 3.14V on the radio module.

In this board (with resistor divider) I have the error:

FAILURE 
/root/Arduino/libraries/arduino-lmic-master/src/lmic/radio.c:689

but not in the first one, with level shifter, that work fine.

This error means SPI failure on reading the RegVersion register, at least in my case.
To see the signal in the SPI interface on the radio I comment out this line in radio.c:

//some sanity checks, e.g., read version number    
//u1_t v = readReg(RegVersion);

and add the follow line to continuosly send packed over SPI interface and check the lines with an oscilloscope:
u1_t v = 0;while (1) {v = readReg(RegVersion);}		

Result: The MOSI, NSS; SCK signals are identical on the 2 boards (except for little difference in the shape of the resistor divider board, but not important at 4 Mhz), but there isn't any signal on the
MISO line, because the RST line is always LOW in the board with resistor divider (and HIGH in the working one with level shifter), and this hold the radio in the reset state.

I investigate in the code, and I found a line (in radio.c)I not understand:

void radio_init () {
    hal_disableIRQs();

    // manually reset radio
#ifdef CFG_sx1276_radio
    hal_pin_rst(0); // drive RST pin low
#else
    hal_pin_rst(1); // drive RST pin high
#endif
    hal_waitUntil(os_getTime()+ms2osticks(1)); // wait >100us
    hal_pin_rst(2); // configure RST pin floating!                  <<<<<<--------- THIS LINE!
    hal_waitUntil(os_getTime()+ms2osticks(5)); // wait 5ms
    opmode(OPMODE_SLEEP);	
    // some sanity checks, e.g., read version number    
    u1_t v = readReg(RegVersion);
    
#ifdef CFG_sx1276_radio
    ASSERT(v == 0x12 );
#elif CFG_sx1272_radio
    ASSERT(v == 0x22);
#else
#error Missing CFG_sx1272_radio/CFG_sx1276_radio
#endif

My question is, why just before to perform a readReg operation the RST pin is set floating?
I suppose that with this pin floating, if I use a level shifter I can have the pin at a high level thanks to the internal logic of the level shifter, but if I use a resistor divider
the 1K8 resistor tie to ground the RST pin and hold the radio in the reset state.

I modified the code above by commenting out the line where RST pin is configured floating and driving the RST pin high before perform the SPI operation, as follow:

void radio_init () {
    hal_disableIRQs();

    // manually reset radio
#ifdef CFG_sx1276_radio
    hal_pin_rst(0); // drive RST pin low
#else
    hal_pin_rst(1); // drive RST pin high
#endif
    hal_waitUntil(os_getTime()+ms2osticks(1)); // wait >100us
    //hal_pin_rst(2); // configure RST pin floating!
    hal_waitUntil(os_getTime()+ms2osticks(5)); // wait 5ms
	hal_pin_rst(1);
    opmode(OPMODE_SLEEP);	
    // some sanity checks, e.g., read version number    
    u1_t v = readReg(RegVersion);
    
#ifdef CFG_sx1276_radio
    ASSERT(v == 0x12 );
#elif CFG_sx1272_radio
    ASSERT(v == 0x22);
#else
#error Missing CFG_sx1272_radio/CFG_sx1276_radio
#endif

And now all work fine on both the boards, and solved the error.

I hope this can help someone!

bye!
Cristiano

matthijskooijman, 2020-06-21:
> My question is, why just before to perform a readReg operation the RST pin is set floating?

The idea is that the RST pin has an internal pullup, so you just let the pullup set the right state. I guess manually driving the RST pin could also work (though it would need to be inverted for 1276), not sure why the original authors chose to do it like this.

cyberman54, 2020-06-21:
This is intentionally, since hardware datasheet of SX127x says pins shall left floating. Look here: https://github.com/mcci-catena/arduino-lmic/issues/579

cristianomazzuoli, 2020-06-21:
Hi all and thanks for your help.
Yes, as said from  cyberman54 and  terrillmoore in the linked post, the problem is the resistor divider, due to the 2 k resistor to gnd that hold the chip in the RST state.
The RFM96 datasheet is clear about this and it tell to left the RST floating for normal operation and drive LOW only for reset, so as terrillmoor says, the code is right and respect the datasheet.
I understand that is always better use a level shifter rather than a resistor divider to adapt different voltage levels, but this is normal pratice especially for low speed SPI comunications.
However I took into consideration only the delay introduced by the resistive divider in the waveform and not other complications like this.
So, if I want use resistor divider instead of ic level shifter, I think is right to comment out the line where the RST line is set to floating and manually drive this pin high.
Another better solultion that respect the datasheet specs, would be not use a resistive divider in the RST line but only a 1k serie resistor between arduino pin and RFM96 RST pin, to avoid any damage if accidentally the arduino pin is set to high, and left the code as the original. So I can drive low RST pin when I want reset the radio, but the floating state during operation is respected and not affected.
I make this test and confirm you later if this work fine.
Thank all for your time and for your work.
Have a nice day!
Cristiano

cristianomazzuoli, 2020-06-21:
Yes, I confirm you that by removing the 1K8 resistor to ground in the resistive divider and left only the 1K serie resistor in the RST line from arduino to radio all work fine with the original code.
Thanks!

guljanjua, 2021-09-17:
Hi,

I hope you are doing good. I fixed this issue. Please replace this:
`
#ifdef CFG_sx1276_radio
    ASSERT(v == 0x12 );
#elif CFG_sx1272_radio
    ASSERT(v == 0x22);
#else
#error Missing CFG_sx1272_radio/CFG_sx1276_radio
#endif
`

with this:
`
#ifdef CFG_sx1276_radio
    if(v != 0x12 )
        return 0;
#elif CFG_sx1272_radio
    if(v != 0x22)
        return 0;
#else
#error Missing CFG_sx1272_radio/CFG_sx1276_radio
#endif
`

in radio.c line 688.

KaiserFels, 2021-12-04:
@guljanjua I tried this but I get an error on line 737. Do you know how can I solce it?

guljanjua, 2021-12-04:
Hi @KaiserFels ,

I hope you are doing good. Well I solved this error but still library was not working and was not uploading data to the thingsnetwork. So I ended up using this library: [https://github.com/mcci-catena/arduino-lmic](url)

One more thing, if you are using Uno and adding GPS or any extra library that will use memory so a stability problem may occur and Arduino will send 2-3 LoRa packets and than it stop sending data. You may need to go for arduino with more memory. 

I hope it will be helpful for you.

KaiserFels, 2021-12-05:
I am using TTGO v2.1 module. I used the library you posted and after selecting EU868 as the Region, I got the following error. "Error: MCCI_LoRaWAN_LMIC_library-3.0.99\src\lmic\oslmic.c:53"
Now I checked the pins in getpinmap_ttgo_lora32_v2.1.cpp and it was correct. I can't find the solution.

guljanjua, 2021-12-06:
Well I am sorry I have not tried this library on esp32. Maybe try changing pin to another pin and see if it works.

[1 more reply]