matthijskooijman/arduino-lmic #41: MAC Command Link_Adr_Req successful

cyberman54, 2016-09-03:
I am using an Ideetron Nexus with this LMIC Stack in German Digimondo LoRaWAN with SF7BW125. After i while i found the Nexus transmitting with SF8BW125 and was surprised. After some analysis i found that it obviously has beend switched by network MAC commands - that means, MAC commands work with this LMIC stack!

This is what i see on Digimondo side:

{
  "port": 0,
  "pending": false,
  "mtype": "unconfirmed_data_down",
  "mic_pass": true,
  "major": 0,
  "mac_cmds": [
    {
      "type": "link_adr_req",
      "tx_power": 0,
      "number_of_transmissions": 1,
      "data_rate": 3,
      "channel_mask_ctrl": 6,
      "channel_mask": 65535
    }
  ],
  "fopts_len": 5,
  "fcnt": 229,
  "dir": "down",
  "dev_addr_hex": "17a98428",
  "adrackreq": false,
  "adr": false,
  "ack": false
}

matthijskooijman, 2016-09-19:
Thanks for confirming. I'm closing this issue, though, since I don't think there is anything left to do?

closed 2016-09-19 by matthijskooijman

cyberman54, 2016-10-20:
After i while using this Code if found, that on Digimondo Network ADR Requests to Downgrade datarate work, but if once Downgrade, MAC commands to Upgrade the Connection seems not work.

my node started with SF7, was then Downgraded by ADR MAC commands to SF8, SF9 and finally SF12. Now it remains in SF12, eben if Network sends ADR to Lift it up again.

I am not 100% sure if this is a Network or node issue.