matthijskooijman/arduino-lmic #42: ping / LMIC_setPingable

lukastheiler, 2016-09-04:
Hi, 

i'm sending messages down to the node, and it works fine: when the node sends a message, the EV_TXCOMPLETE event reports the downlink message after sending something.  

Now I've read about LMIC_setPingable(x) that sends a ping every 2^x seconds. I've put the call right before the main os_runloop(); .This should then trigger a EV_RXCOMPLETE event, with the downlink message. However, I didn't manage to get that running. 

What am I missing? Do i need to send the ping manually somehow? 

Thanks! Lukas

matthijskooijman, 2016-09-04:
I think "Pingable" only applies to class B devices, which requires gateways to send beacon frames. I'm not sure if any networks currently do this, TTN certainly does not.

lukastheiler, 2016-09-05:
Ah, thanks a lot for clairifying!

closed 2016-09-05 by matthijskooijman