matthijskooijman/arduino-lmic #46: D0/D1/D2 pull-up, pull-down or left floating?

MarsWarrior, 2016-10-01:
I'm currently using the same PCB for both RFM69 and RFM9x Lora chips.

In the case of the RFM69 the D0 pin is the interrupt line, and therefore is connected to a pull-up resistor.
The other two pins, D1 and D2 are left floating (no pull-up/down).

In order to use the RFM9x chip and the arduino-lmic library, should I change the pull-up on D0 and add pull-downs on D1/D2, or will it work the way it is now?

matthijskooijman, 2016-12-06:
I believe no pullups are needed, since the DIO pins on the SX127x chips (which are on the RFM boards) have push-pull drivers (e.g. not open-collector, low-or-floating, but fixed high or low voltage). Having a pullup won't hurt either, though.

However, by default you will need to connect at least DIO0 and also DIO1 and/or DIO2 (depending on whether you use LoRa or FSK). There is some code floating around to work without any DIO connections, but that's not very polished yet.

matthijskooijman, 2017-01-19:
I'm assuming this is resolved, so I'm closing this issue.

closed 2017-01-19 by matthijskooijman

hallard, 2017-01-19:
Guys,
take care, as far has I remember on RF95 (so LMIC) DIO line are active high, so a pull up may hurt, in this case it should be a pull down.
But @matthijskooijman is right, no need to add any pull/up down on DIO line. I never added any pull UP/DOWN on RFM95 modules, and it works on Arduino/WeMos/Esp8266/RasPI/CHIP

trlafleur, 2017-01-19:
You need a high value weak pull down with some processors, M0 device has selectable pull down on all io lines.   I use 47k

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

MarsWarrior, 2017-06-26:
A bit late, but thanks for all the answers!
It seems to work without any pull ups. Hallards solution with a pull down & the three diode "OR" on the DIO lines works for me & do save a few I/O lines!

I have added a solder jumper on the D0 line, so I can use both RFM95 and RFM69.