matthijskooijman/arduino-lmic #49: Arduino Mega Cant Send Data To TTN Server!

KhaledAljehani, 2016-10-12:
Hi,

I am using Arduino Mega + Adafruit RFM95 (SX1276). The code is working, but I do not know how to send data to The Things Network server, and view them. 

This is the console output:
[image]

I really need your help in how to do TTN server configuration.

Thanks in advance.

jmidd, 2016-10-12:
You shouldn't expect support for TTN to be provided here - see https://www.thethingsnetwork.org and register on the forum, its all there !

alex1sh, 2016-10-14:
Did you setup the keys? Please check out this guide:
https://github.com/lukastheiler/ttn_moteino#moteino-lmic-and-otaa-walkthrough-wip

KhaledAljehani, 2016-10-14:
First of all, I really appreciate your help @alex1sh @jmidd 

I set-up the keys correctly, but I could not receive data in the TTN server. By the way, I did set-up a gateway in addition to the Arduino, and honestly I do not know if the Arduino code will send data to the server or not. 

So, shoud I follow this plan to send data to the server? The plan:
1. Set-up a gateway
2. Connect the gateway to TTN server
3. Send the data in Arduino to the gateway.
4. Then, it will be shown in the server.

Please, provide me with any useful source.

Thanks in advance

gizmocuz, 2016-10-15:
Just to be sure...:

You reversed the keys for the  APPEUI and DEVEUI ?:

{0x12, 0x34, 0x56, 0x78, 0x9A, 0xBC, 0xDE, 0xF0}

becomes

{0xF0, 0xDE, 0xBC, 0x9A, 0x78, 0x56, 0x34, 0x12}

Also when NOT using OTAA (thus ABP), there is a thing with the frame counter, it should always increment.
If you restart your node, and 1 frame was already received, TTN will expect the next packet to send 2, but as the node is restarted it will send 1.
You can enable 'relaxed' frame counting in the web gui:
[image]

I would always start by making a ABP sketch, when you got this working, then start experimenting with ATAA

I hope for you that you have a 'real' gateway arround that is not yours.
Or do you have a real gateway, that listens to at all frequencies ?
If you created your own single channel gateway, then all your communication will have to go via 1 frequency.

Rest to say that this library is 100% working, and the best place to ask for help is on the forum of TTN

KhaledAljehani, 2016-10-15:
Yes, I set-up all keys correctly.

There is one thing I want to make sure of, which is: is this code that I am suing (ttn-abs.ino) sends data directly to the server? Or I should send it to a gateway like raspberry pi?

This is the most importatn thing I want to know

I really apperciate your help @jmidd @alex1sh @gizmocuz @matthijskooijman

alex1sh, 2016-10-15:
On the keys, please use ttnctl with the following parameters:

ttnctl devices info DEVEUI --lmic

copy and paste the keys (APPEUI and DEVEUI are in tlittle endian format) into the sketch.

Unfortunately, I am having difficulty myself with OTAA running it on my RPi3

KhaledAljehani, 2016-10-15:
But the sketch requires only these keys: NWKSKEY, APPSKEY, and DEVADDR. It does not ask for App EUI.

alex1sh, 2016-10-15:
OK,. you have ABP - this is your next step OTAA

KhaledAljehani, 2016-10-15:
Excuse me, do you have an answer for this question?

"There is one thing I want to make sure of, which is: is this code that I am suing (ttn-abs.ino) sends data directly to the server? Or I should send it to a gateway like raspberry pi?"

Thanks in advance

alex1sh, 2016-10-15:
it depends. Do you have LoRaWAN gateway in your area? If the answer if no, In this case you have to come up with your own GW

KhaledAljehani, 2016-10-15:
No, I do not have a gateway in my area. Could you please suggest me a good source to start with?

alex1sh, 2016-10-15:
You really have to start reading TTN forum. Things change very quickly and there is a lot of information to digest.
 
Your best bet would be a single channel GW - there is a tread on the forum.  I have one of these - https://github.com/hallard/WeMos-Lora. 

Alternatively, you can assemble a proper GW with IC880 (multiple freq. and 100% LoRaWAN compatible). Other alternative would be to wait for the official TTN GW, which is due to come out in Nov 2016

KhaledAljehani, 2016-10-15:
Thanks. I really appreciate your help.

alex1sh, 2016-10-15:
This would be a good starting point for you - https://github.com/ttn-zh/ic880a-gateway/wiki
Please note that this is a completely DIY project.

KhaledAljehani, 2016-10-15:
OK