matthijskooijman/arduino-lmic #55: Downlink not working on Luigino (Arduino) ATMEGA328p 8Mhz

kenshiro75, 2016-11-11:
hi,
i' m using your LMIC library on my node which is a Luigino equipped with Atmega328p 8Mhz 3.3v and a RFM95W (SX1276) : i can send to my gateway the message and look at it in the TTN payload log but when i try a downlink in the gateway's log i can see the downlink has been sent but there's no hope to see it as received in the node's logs...
what should i tune/modify to get it working ?
eventually what should i check into the gateway to understand if the downlink has really been delivered to the node ?
i read in the issues that with Arduino's boards there are different problems regarding clock/SPI sync or settings...

thank you for your time.

Ken

please note that i tried another node equipped with the RN2483 and i can get the downlink from the TTN without any problem .

matthijskooijman, 2016-11-11:
One thing you can quickly try, is to let LMIC compensate for an inaccurate clock. Just add this to your setup somewhere:
```
  // Let LMIC compensate for +/- 1% clock error
  LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);
```

This lets LMIC make the RX windows longer and start sooner so it will catch a downlink even when the clock runs up to 1% daster or slower.

kenshiro75, 2016-11-11:
mmm... i've added your code but still no success.. here the node's log :

```
Packet queued
5077310: RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=0
5139555: RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
5403488: EV_TXCOMPLETE, time: 86
EV_TXCOMPLETE (includes waiting for RX windows)
5404702: engineUpdate, opmode=0x800
6029702: engineUpdate, opmode=0x808
6030675: TXMODE, freq=868100000, len=26, SF=7, BW=125, CR=4/5, IH=0
Packet queued
6095067: RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=0
6157310: RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
6405080: EV_TXCOMPLETE, time: 102
EV_TXCOMPLETE (includes waiting for RX windows)
6406357: engineUpdate, opmode=0x800
```

this is part of the settings in the node's sketch :

```
const unsigned TX_INTERVAL = 10;
// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = LMIC_UNUSED_PIN,
    .dio = {2, 3, 4},
};
```

this is the setup section:

```
void setup() {
    
    Serial.begin(9600);
    Serial.println(F("Starting"));
    
    #ifdef VCC_ENABLE

    pinMode(VCC_ENABLE, OUTPUT);
    digitalWrite(VCC_ENABLE, HIGH);
    delay(1000);
    #endif

    os_init();

    LMIC_reset();

    #ifdef PROGMEM

    uint8_t appskey[sizeof(APPSKEY)];
    uint8_t nwkskey[sizeof(NWKSKEY)];
    memcpy_P(appskey, APPSKEY, sizeof(APPSKEY));
    memcpy_P(nwkskey, NWKSKEY, sizeof(NWKSKEY));
    LMIC_setSession (0x1, DEVADDR, nwkskey, appskey);
    #else
    // If not running an AVR with PROGMEM, just use the arrays directly
    LMIC_setSession (0x1, DEVADDR, NWKSKEY, APPSKEY);
    #endif

    #if defined(CFG_eu868)
    
    LMIC_setupChannel(0, 868100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(1, 868300000, DR_RANGE_MAP(DR_SF12, DR_SF7B), BAND_CENTI);      // g-band
    LMIC_setupChannel(2, 868500000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(3, 867100000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(4, 867300000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(5, 867500000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(6, 867700000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(7, 867900000, DR_RANGE_MAP(DR_SF12, DR_SF7),  BAND_CENTI);      // g-band
    LMIC_setupChannel(8, 868800000, DR_RANGE_MAP(DR_FSK,  DR_FSK),  BAND_MILLI);      // g2-band
    for (int i=1; i<16; i++) { LMIC_disableChannel(i); }

    #elif defined(CFG_us915)

    LMIC_selectSubBand(1);
    #endif
    
    // Disable data rate adaptation
    LMIC_setAdrMode(0);

    // Disable link check validation
    LMIC_setLinkCheckMode(0);
 

    // TTN uses SF9 for its RX2 window.
    LMIC.dn2Dr = SF9;

    // Set data rate and transmit power for uplink (note: txpow seems to be ignored by the library)
    LMIC_setDrTxpow(DR_SF7,14);

    // Let LMIC compensate for +/- 1% clock error
    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);
   
    // Start job
    do_send(&sendjob);
    
}

```

as you can see i'm using only the SF7 on the freq 868.1 : i can clearly see the downlink that the gateway is trying to send back after having received the node's message (from the gateway's logs)

```
PKT_PULL_DATA request: <12> 1:9D:5D:2:5C:CF:7F:FF:FF:5:FB:D1:
sendUdp 1: sent 12 bytes
sendUdp 2: sent 12 bytes
PKT_PULL_ACK:: size 4 From 192.168.2.112, port 1700, data: 1:9D:5D:4:
PKT_PULL_ACK:: size 4 From 40.114.249.243, port 1700, data: 1:9D:5D:4:
receivePacket:: LoRa message ready
Packet RSSI: -65 RSSI: -109 SNR: 9 Length: 26 -> 40 CA 68 FC 2B 0 1 0 1 B6 F6 8 76 E8 61 6A 69 E8 8A F3 3A 55 E0 FC F8 C7 
RXPK:: {"rxpk":[{"tmst":690312645,"chan":0,"rfch":0,"freq":868.099975,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":9,"rssi":-65,"size":26,"data":"QMpo/CsAAQABtvYIduhhamnoivM6VeD8+Mc="}]}
sendUdp 1: sent 207 bytes
sendUdp 2: sent 207 bytes
PKT_PUSH_ACK:: size 4 From 192.168.2.112, port 1700, token: 4044

PKT_PUSH_ACK:: size 4 From 40.114.249.243, port 1700, token: 4044

{"txpk":{"imme":false,"tmst":691312645,"freq":868.099,"rfch":0,"powe":14,"modu":"LORA","datr":"SF7BW125","codr":"4/5","ipol":true,"size":17,"data":"YMpo/CsABwABU96Ohz3keQY="}}
data: YMpo/CsABwABU96Ohz3keQY=
txLoraModem::  powe: 14, freq: 868100000, crc: 0, iiq: 0X40
setFreq using: 868100000
start: 690970135, end: 691312645, waited: 342510, delay=0
setFreq using: 868100000
Request::  tmst=691312645 wait=373675
 strict=1 datr=SF7BW125
 freq=868100000 ->868100000
 sf  =7 ->7 modu=LORA powe=14 codr=4/5
 ipol=1

60:CA:68:FC:2B:0:7:0:1:53:DE:8E:87:3D:E4:79:6:
PKT_PULL_ACK:: tmst=691376007
PKT_PULL_RESP:: size 179 From 40.114.249.243, port 1700, data: txpk..
```

thank you
Ken

matthijskooijman, 2016-11-16:
Hm, I can't find an obvious fault here. The gateway log indeed shows the packet being scheduled and sent exactly 1 second after txend.

One thing that is slightly weird is that the time from the TXMODE to the first RXMODE in your node log is (6095067-6030675)*16=1030272us, which is slightly more than 1s. The timestamps in the log output are approximate, so it might not be your problem, but perhaps it is. You could try increasing the RX window further, by adapting for +/-10% of clock inaccuracy:

    LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);

hichem5, 2016-11-24:
hi,
I have the same problem with a sheild darginho (sx1276) connected to an arduino MEGA, I do not receive messages sent by the Gateway.[ Forum TTN](https://www.thethingsnetwork.org/forum/t/node-not-receiving-downlink-message/4161)

The Gateway sends a message thanks to the script Node.js that I execute as you see the result.
```
>  INFO: [down] for server router.eu.staging.thethings.network PULL_ACK received in 67 ms
>  INFO: [down] for server router.eu.staging.thethings.network serv_addr[ic]PULL_RESP received :)
>  INFO: [down] a packet will be sent on timestamp value 215351708
>  INFO: [down] for server router.eu.staging.thethings.network PULL_ACK received in 68 ms
>  INFO: [down] for server router.eu.staging.thethings.network PULL_ACK received in 98 ms
>  
>  ##### 2016-11-24 13:27:14 GMT #####
>  ### [UPSTREAM] ###
>  # RF packets received by concentrator: 1
>  # CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
>  # RF packets forwarded: 1 (29 bytes)
>  # PUSH_DATA datagrams sent: 2 (479 bytes)
>  # PUSH_DATA acknowledged: 0.00%
> ### [DOWNSTREAM] ###
>  # PULL_DATA sent: 3 (100.00% acknowledged)
>  # PULL_RESP(onse) datagrams received: 1 (177 bytes)
>  # RF packets sent to concentrator: 1 (17 bytes)
> # TX errors: 0
>  ##### END #####
```

But the arduino monitor running ttn-abp does not show a received message.

```
 Starting
RXMODE_RSSI
203: engineUpdate, opmode=0x808
716: TXMODE, freq=868100000, len=29, SF=7, BW=125, CR=4/5, IH=0
Packet queued
64139: RXMODE_SINGLE, freq=868100000, SF=7, BW=125, CR=4/5, IH=0
126383: RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
EV_TXCOMPLETE (includes waiting for RX windows)
No Things Received 
403760: engineUpdate, opmode=0x800
2278761: engineUpdate, opmode=0x808
2279286: TXMODE, freq=868300000, len=29, SF=7, BW=125, CR=4/5, IH=0
Packet queued
2343414: RXMODE_SINGLE, freq=868300000, SF=7, BW=125, CR=4/5, IH=0
2405658: RXMODE_SINGLE, freq=869525000, SF=9, BW=125, CR=4/5, IH=0
EV_TXCOMPLETE (includes waiting for RX windows)
No Things Received 
```
Please what can I change to receive the downlink message?