matthijskooijman/arduino-lmic #57: Easy way to restart arduino on `FAILURE`?

ubergesundheit, 2016-11-19:
How can I make sure my arduino restarts after the library encountered a `FAILURE` condition? I know of the `asm volatile (" jmp 0");`, but how can I integrate it into the library?

matthijskooijman, 2016-11-19:
I think the most reliable way would be to use watchdog, which will cause a real reset (but that requires a bootloader that clears `MCUSR` as well).

In any case, you would need to modify the `hal_failed()` function in `hal.cpp`. Perhaps it makes sense to make that function weak, so you can just define your own in the sketch to override it.

ubergesundheit, 2016-11-19:
Do you have any tips or pointers for me how I can make the `hal_failed()` function weak? My C/C++ is really bad.

matthijskooijman, 2016-11-19:
Add __attribute__((weak)) in its declaration IIRC (but Google should help with that).

Oliv4945, 2017-01-20:
@matthijskooijman I think it would be nice to declare it as weak as default in the library.

geppoleppo, 2019-10-22:
> Do you have any tips or pointers for me how I can make the `hal_failed()` function weak? My C/C++ is really bad.

> Do you have any tips or pointers for me how I can make the `hal_failed()` function weak? My C/C++ is really bad.
Hi Gerald, do you have solved the reset function in FAILURE condition?

ubergesundheit, 2019-10-22:
Sorry @geppoleppo, I didn't solve this

Roberto6969, 2020-02-23:
Old thread but still open: it would be great if someone will give advice how to implement wdt with LMIC. Maybe with an example?