matthijskooijman/arduino-lmic #58: First message lost after reset

pablomargareto, 2016-11-19:
Hello,

I am using this library with my NodeMCU (ESP8266).

Everything works fine if I send messages periodically. The problems comes when I reset twice.

Example:
- I send periodically and I reset => No problem, the first message is sent
- I reset again => The first message is lost (I can see the counter in TTN)

Once I receive a new message again, the cycle repeats.

I am using the sub band 1 in the US (through select sub band). Can anyone give me some hint about what I am doing wrong?

Thanks!

Oliv4945, 2016-11-20:
Hi,

This is not a bug : there is a counter (lmic.fCnt) inside the stack in order to help the server to detect uplink resend.  So with the reset you send a packet twice with 0, the second is dropped. 
If you are using TTN try to declare your device with the option `--relax-fcnt` or use OTAA.

matthijskooijman, 2016-11-20:
Yup, it is as Oliv4945 explains. You should either use OTAA, or save the frame count. As a workaround you can use the frame count relaxation, but I suspect that you are already doing this and this is exactly what triggers this behaviour (since TTN sees frame counter 0 twice and assumes that the second one is a duplicate). In any case, not a LMIC problem I believe, so I'm closing this issue.

closed 2016-11-20 by matthijskooijman

Oliv4945, 2016-11-20:
> or save the frame count
I would not do that way, or saving all lmic structure : when reseting the stack will use default parameters, and the network will not send its current set as it will not know that the node has restarted