matthijskooijman/arduino-lmic #59: Dragino shield V1.3 : Pin mapping

urs8000, 2016-12-25:
for the documentation
// Pin mapping  for Dragino Board Version 1.3   (open J_DIO5)
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 9,
    .dio = {2, 6, 7},         // Specify pin numbers for DIO0, 1, 2
                                   // connected to D2, D6, D7 
};
not tested with LoRa/GPS shield!

kazi93, 2017-04-05:
ANY INFO: Dragino Board Version 1.4 ? Thanks

henri98, 2018-01-13:
The pin mapping for dragino board version 1.4 are the same.

MatGit5, 2018-03-18:
I can confirm it works with Dragino Board Version 1.4

Sasch600xt, 2018-08-28:
may i ask it will also work with the Arduino Mega 2560 ?
i did try since 2 days an i am not sure i am sending data.....gateway is recieving nothing at all.
i dont have an UNO or an Feather......
Do you know there are special needs for the Mega ?
Thank you so much for your help

Roberto6969, 2018-08-28:
Actually not - it works fine with 2560 Core also: https://www.aliexpress.com/item/Mega2560-CORE-mini-2560-Arduino-compatible-3-3V-5V-for-Arduino-mega-2560/32603415477.html
Just be carefull with pin mappings - there is many more pins on 2560

Sasch600xt, 2018-08-28:
well....i am not a pro right now.
"Just be carefull with pin mappings - there is many more pins on 2560" 

ummm......i have no idea how to do the pin mapping.....
I know i am asking for very much, but could one of you maybe have a look at the Arduino Mega pinout and tell me what to do ?

I think we talking about this section : 
`// Pin mapping
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 9,
    .dio = {2, 6, 7},
};

Thank you so much for your help !
Sascha`

Roberto6969, 2018-08-28:
Here is my mappings:
const lmic_pinmap lmic_pins = {
  .nss = 53,  // 10 Connected to pin D10
  .rxtx = LMIC_UNUSED_PIN,  // 0 For placeholder only, Do not connected on RFM92/RFM95
  .rst = 30,   // 9 Needed on RFM92/RFM95? (probably not)
  .dio = {5, 6, 7}, // Specify pin numbers for RFM95 DIO0, 1, 2

In my case i use only RFM95 module (not Dragino) connected directly to 2560 Core. Used pins between RFM95 and 2560 Core are:
RFM95 - 2560 Core name and pin number
   -----------------
   MISO  - PB3 22   
   MOSI  - PB2 21   
   SCK    - PB1 20   
   NSS    - PB0 19   
   RESET - RESET 30 
   DIO0  - PE3 5    
   DIO1  - PH3 15   
   DIO2  - PH4 16   

If you use Dragino this will help you: http://wiki.dragino.com/index.php?title=Lora_Shield

mantoles, 2020-10-18:
Can anyone tell me what the dragino pin mappings are with Leonardo?

I tried the following and almost bricked my leonardo.

const lmic_pinmap lmic_pins = {
 .nss = 10,
 .rxtx = LMIC_UNUSED_PIN,
  .rst = 9,
 .dio = {2, 6, 7},
 };

matthijskooijman, 2020-10-18:
@mantoles, the Leonardo should have the same pin numbers for digital I/O as the Uno, so that pin mapping should be correct. However, the Leonardo does *not* have SPI pins available on digital pin 11/12/13, only on the 2x3pin ICSP connector. I haven't checked the dragino board, but if that board assumes SPI on 11/12/13 and does not have the 2x3 pin connector, you'll have to reroute the SPI connections somehow (i.e. add wires, maybe cut pins or traces on the Dragino).

See also https://www.arduino.cc/en/reference/SPI

mantoles, 2020-10-18:
@matthijskooijman Much appreciate your response.  Now it makes more sense.  The leonardo has the 2x3 ISP connector so it should be connected to the spi bus directly.

Phairou18, 2023-06-01:
Good morning,
I am using an Arduino Mega 2560 board with a LoRa Shield having the RFM95W radio module. However I use LMIC to send my data to a Dragino LPS8N gateway.
And when I upload my program to the Arduino board nothing happens and I get the error displayed on the serial monitor:
Arduino\libraries\LMIC-Arduino\src\lmic\radio.c:689

I thought at first it was a pin mapping problem but I still have the same problem.
Help me please.
Here is my test code on arduino:

const int test =13;
//---------------------------------------------------

#include <lmic.h>
#include <hal/hal.h>
#include <SPI.h>

// LoRaWAN NwkSKey, network session key
// This is the default Semtech key, which is used by the early prototype TTN
// network.
static const PROGMEM u1_t NWKSKEY[16] = { 0x91, 0xA4, 0xAE, 0x05, 0x61, 0x89, 0x16, 0xCF, 0x09, 0x73, 0x51, 0x70, 0x19, 0xD1, 0xC3, 0xE2 };

// LoRaWAN AppSKey, application session key
// This is the default Semtech key, which is used by the early prototype TTN
// network.
static const u1_t PROGMEM APPSKEY[16] = { 0x79, 0xDB, 0xD8, 0xE4, 0x98, 0xF8, 0x19, 0xE1, 0xA6, 0x52, 0x5D, 0x4F, 0x79, 0x43, 0x71, 0x5D };

// LoRaWAN end-device address (DevAddr)
static const u4_t DEVADDR = 0x010C2D70; // <-- Change this address for every node!

// These callbacks are only used in over-the-air activation, so they are
// left empty here (we cannot leave them out completely unless
// DISABLE_JOIN is set in config.h, otherwise the linker will complain).
void os_getArtEui (u1_t* buf) { }
void os_getDevEui (u1_t* buf) { }
void os_getDevKey (u1_t* buf) { }

static uint8_t mydata[] = "Hello, world!";
static osjob_t sendjob;

// Schedule TX every this many seconds (might become longer due to duty
// cycle limitations).
const unsigned TX_INTERVAL = 10;

// Pin mapping
const lmic_pinmap lmic_pins = {
.nss = 10,
.rxtx = LMIC_UNUSED_PIN,
.rst = 9,
.dio = {2, 6, 7},
};

void onEvent (ev_t ev) {
Serial.print(os_getTime());
Serial.print(": ");
switch(ev) {
case EV_SCAN_TIMEOUT:
Serial.println(F("EV_SCAN_TIMEOUT"));
break;
case EV_BEACON_FOUND:
Serial.println(F("EV_BEACON_FOUND"));
break;
case EV_BEACON_MISSED:
Serial.println(F("EV_BEACON_MISSED"));
break;
case EV_BEACON_TRACKED:
Serial.println(F("EV_BEACON_TRACKED"));
break;
case EV_JOINING:
Serial.println(F("EV_JOINING"));
break;
[... 123 lines cut ...]

// Disable link check validation
LMIC_setLinkCheckMode(0);

// TTN uses SF9 for its RX2 window.
LMIC.dn2Dr = DR_SF9;

// Set data rate and transmit power for uplink (note: txpow seems to be ignored by the library)
LMIC_setDrTxpow(DR_SF7,14);

// Start job

do_send(&sendjob);
}

//----------------------------------------------------------------------
void loop() {
// put your main code here, to run repeatedly:
os_runloop_once();
digitalWrite(test,HIGH);
delay(1000);
digitalWrite(test,LOW);
delay(1000);

}