matthijskooijman/arduino-lmic #60: Arduino uno with LMIC lib. assert failure

pluss2, 2016-12-28:
**The goal what I want is OTAA(over the air activation) communication with Arduino(or raspberry pi 3) - Gateway(raspberry pi 3 or arduino) - TTN(the thing network).**

_Ready to execute code.._
**1.** Arduino Uno R3, computer-arduino serial cable
**2.** Arduino shield(https://www.cooking-hacks.com/documentation/tutorials/multiprotocol-shield-connect-two-xbee-connector-arduino-raspberry-pi-galileo), sx1272 chip
**3.** LMIC library Version 1.5
**4.** Upload library to Arduino(IDE V1.6.11) sketch library.
**5.** Not modify pin map struct in raw.ino
**6.** [Circuit image](https://www.cooking-hacks.com/media/cooking/images/documentation/tutorial_kit_lora/lora_arduino_big.jpg)

At first, i want to phy level communication(raw.ino in lmic library)test with just one arduino. So i modify some files like below

config.h
//#define CFG_eu868 1
//#define CFG_sx1276_radio 1
#define CFG_us915 1
#define CFG_sx1272_radio 1
#define DISABLE_INVERT_IQ_ON_RX

hal/hal.cpp
static const SPISettings settings(10E6, MSBFIRST, SPI_MODE0)
// static const SPISettings settings(1E6, MSBFIRST, SPI_MODE0)
// static const SPISettings settings(4E6, MSBFIRST, SPI_MODE0)

Question 1. I tried all the comments, but i get the following error. how can i fix error ?
Starting
FAILURE 
C:\Users\Jaeho Shin\Documents\Arduino\libraries\arduino-lmic-master\src \lmic\radio.c:702 

Question 2. is this [link](https://github.com/TheThingsNetwork/ttn-gateway-connector) correct gateway source code ?

Oliv4945, 2016-12-28:
@pluss2 
> Question 1.

There is no assertion in radio.c [L702](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/radio.c#L702) in master, did you modify the code ? Could you please show your line 702 ?
I assume that it is the assertion about chip version, could you please try to print the variable `v` [L687](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/radio.c#L687) ? Did you double checked `lmic_pins` declaration and jumpers on your shield ?

pluss2, 2016-12-28:
Sorry about that.
The error appear at line [L691](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/radio.c#L691).

So, when I check the value v, v is 82. what is the problem ? and what is number 82 means ?

Oliv4945, 2016-12-28:
>Q1. what is the problem ? and what is the number 82 means ?

I do not know. It should be 0x22 for the SX1272. Do you have an oscilloscope or logic analyser to hook up and verify SPI communication ?

>Q2. is lmic_pins value right?
>Q3. should i control Multiprotocol shield (2.1)?

`lmic_pins` depends of your shield, it seems that yours have an I/O expander so you should ask Libellium about that.

trlafleur, 2016-12-28:
If you look at the data sheet for the radio, each radio model has a unique
ID

SX1276 is 0x12 and SX1272 is 0x22   so, it checking to make sure it has a
correct radio attach....

if you have a RFM9x series radio, it should report one of these... unless
their a new variant out there...