matthijskooijman/arduino-lmic #61: What is the pin map of Arduino uno & sx1272 with LMIC library.

pluss2, 2016-12-29:
I have Arduino Uno & sx1272 chip.

I want to upload raw.ino. But i don't know the way connect arduino uno & sx1272 direct (**PIN MAP**)

anyone know about arduino uno & sx1272 connect directly ??

Oliv4945, 2016-12-29:
It is written in main readme. Be carrefull to have a level shifter or 3.3v Arduino

pluss2, 2016-12-30:
My code is at [link](https://github.com/matthijskooijman/arduino-lmic/issues/60)

And i wired like image.
[image]
[image]

Also [this](http://cpham.perso.univ-pau.fr/LORA/resources/pinout-xbee-spi.png) is reference of sx1272 chip and [this](http://icircuit.net/wp-content/uploads/2014/09/Arduino-uno.png) is reference of arduino uno pin map

But, it doesn't work, ASSERT ERROR (i check v = 0).

pluss2, 2016-12-30:
I solve this problem with modify my lmic_pins structure like below.
[image]

But now i had new error.
[image]
This is related with [L523](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/radio.c#L523).

i just want to P2P communicate so how can i modify the opmode ?

trlafleur, 2017-01-01:
IF you want Peer-2-Peer, the LoRaWan (IE: LMiC) stack is NOT what you want...
Its designed for node to server communication as use in many iot system.

You should be looking at:
~~~
http://www.airspayce.com/mikem/arduino/RadioHead/
~~~

pluss2, 2017-01-02:
I really want to communicate node with server.
But, lmic library porting is first. so i had to port lmic library at node (arduino uno).
Also, lmic library has peer to peer communication(raw.ino).
And [this](http://cpham.perso.univ-pau.fr/LORA/RPIgateway.html) is what i do.