matthijskooijman/arduino-lmic #63: How can i modify raw.ino ??

pluss2, 2017-01-04:
I successfully porting raw.ino in Arudino UNO.

Is original raw.ino transmit code, right ?

Then, how do i modify raw.ino to receiving node ?

matthijskooijman, 2017-01-07:
The code already receives when not transmitting, see https://github.com/matthijskooijman/arduino-lmic/blob/master/examples/raw/raw.ino#L10-L12

However, this example isn't really the best way to use the library, for node-to-node communication I believe better libraries are available, such as radiohead.

closed 2017-01-07 by matthijskooijman

pluss2, 2017-01-07:
The reason why i discussed with this issue is both node doesn't receive any data.

Ok, i will find the other way with radiohead. 
Thank you so much,

nafitriaulia, 2018-04-29:
Do you success to communicate node peer to peer?