matthijskooijman/arduino-lmic #67: Where the received data is stored?

ariraj, 2017-01-20:
Hello.

I have been debugging my ATSAMD21G18 & SX1272 setup for quite a while now. I'm using OTAA-example as a reference and I have my own gateway & Amazon AWS setup up and running. Motem joins to my private network flawlessly and also transmits data to the cloud just fine. I also get almost always a response from the cloud to the motem after TXCOMPLETE event.

However, in the OTAA-example received data is not displayed, since it's not implemented. Correct my if I'm doing this wrong:

In switch-case statement under EV_TXCOMPLETE, I just added this line:

Serial.write(LMIC.frame, LMIC.dataLen);

In console I just get character ´ printed. As hex LMIC.frame[0] contains 0x60 and LMIC.frame[1] 0x03. LMIC.dataLen is correct, I'm expecting 2 bytes of data.

Console output is always the same, even though the response sent from the cloud is different.

Any ideas what I'm doing wrong? Is there some other rxbuffer?

Oliv4945, 2017-01-20:
Hi, you should use `lmic.databeg`, see this [line](https://github.com/matthijskooijman/arduino-lmic/blob/90bc0492b632475ae918fc9c65cd5508f1fdb1af/src/lmic/lmic.h#L248).

By the way an update of the example might help others, I try to do a PR

ariraj, 2017-01-23:
Thank you for quick response! I was just assuming the received data starts from the beginning of the array, as it does in raw.ino example.

closed 2017-01-23 by ariraj