matthijskooijman/arduino-lmic #68: Arduino error for SX1276 module

ramesh493, 2017-02-16:
Startting Failure

C:\Users\XXX\Desktop\Sketch\libraries\arduino-lmic-master\src\lmic\radio.c 689

I have tried with 1.6.10, 1.6.11 and latest version of arduino.

radio.c code which i am using 

/*******************************************************************************
 * Copyright (c) 2014-2015 IBM Corporation.
 * All rights reserved. This program and the accompanying materials
 * are made available under the terms of the Eclipse Public License v1.0
 * which accompanies this distribution, and is available at
 * http://www.eclipse.org/legal/epl-v10.html
 *
 * Contributors:
 *    IBM Zurich Research Lab - initial API, implementation and documentation
 *******************************************************************************/

#include "lmic.h"

// ----------------------------------------
// Registers Mapping
#define RegFifo                                    0x00 // common
#define RegOpMode                                  0x01 // common
#define FSKRegBitrateMsb                           0x02
#define FSKRegBitrateLsb                           0x03
#define FSKRegFdevMsb                              0x04
#define FSKRegFdevLsb                              0x05
#define RegFrfMsb                                  0x06 // common
#define RegFrfMid                                  0x07 // common
#define RegFrfLsb                                  0x08 // common
#define RegPaConfig                                0x09 // common
#define RegPaRamp                                  0x0A // common
#define RegOcp                                     0x0B // common
#define RegLna                                     0x0C // common
#define FSKRegRxConfig                             0x0D
#define LORARegFifoAddrPtr                         0x0D
#define FSKRegRssiConfig                           0x0E
#define LORARegFifoTxBaseAddr                      0x0E
#define FSKRegRssiCollision                        0x0F
#define LORARegFifoRxBaseAddr                      0x0F
#define FSKRegRssiThresh                           0x10
#define LORARegFifoRxCurrentAddr                   0x10
#define FSKRegRssiValue                            0x11
#define LORARegIrqFlagsMask                        0x11
#define FSKRegRxBw                                 0x12
#define LORARegIrqFlags                            0x12
#define FSKRegAfcBw                                0x13
#define LORARegRxNbBytes                           0x13
#define FSKRegOokPeak                              0x14
#define LORARegRxHeaderCntValueMsb                 0x14
#define FSKRegOokFix                               0x15
#define LORARegRxHeaderCntValueLsb                 0x15
#define FSKRegOokAvg                               0x16
#define LORARegRxPacketCntValueMsb                 0x16
#define LORARegRxpacketCntValueLsb                 0x17
#define LORARegModemStat                           0x18
#define LORARegPktSnrValue                         0x19
#define FSKRegAfcFei                               0x1A
#define LORARegPktRssiValue                        0x1A
#define FSKRegAfcMsb                               0x1B
#define LORARegRssiValue                           0x1B
#define FSKRegAfcLsb                               0x1C
#define LORARegHopChannel                          0x1C
#define FSKRegFeiMsb                               0x1D
#define LORARegModemConfig1                        0x1D
#define FSKRegFeiLsb                               0x1E
#define LORARegModemConfig2                        0x1E
#define FSKRegPreambleDetect                       0x1F
[... 837 lines cut ...]
// versa, but gateways will not hear other gateways and nodes will not
// hear other nodes. By uncommenting this macro, this inversion is
// disabled and this node can hear other nodes. If two nodes both have
// this macro set, they can talk to each other (but they can no longer
// hear gateways). This should probably only be used when debugging
// and/or when talking to the radio directly (e.g. like in the "raw"
// example).
//#define DISABLE_INVERT_IQ_ON_RX

// This allows choosing between multiple included AES implementations.
// Make sure exactly one of these is uncommented.
//
// This selects the original AES implementation included LMIC. This
// implementation is optimized for speed on 32-bit processors using
// fairly big lookup tables, but it takes up big amounts of flash on the
// AVR architecture.
// #define USE_ORIGINAL_AES
//
// This selects the AES implementation written by Ideetroon for their
// own LoRaWAN library. It also uses lookup tables, but smaller
// byte-oriented ones, making it use a lot less flash space (but it is
// also about twice as slow as the original).
#define USE_IDEETRON_AES

#endif // _lmic_config_h_

sabas1080, 2017-02-16:
The error is in terminal of Arduino IDE or monitor Serial?

Oliv4945, 2017-02-16:
Dear,

What hardware do you use ? If you doubled checked the [pin mapping](https://github.com/matthijskooijman/arduino-lmic#pin-mapping) and [radio declaration](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/config.h#L12), you can slow the SPI communication on this [line](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/hal/hal.cpp#L77) to 2 MHz (2E6) , see #9 for info.

cattvm, 2017-02-17:
Hi Ramesh,

It's the problem with your hardware connection with SX1276, nothing else.
The error point to the line where it can read actual pin. Check power
supply, GND with voltmeter at sx1276.

Can u please share your PIN mapping code with Ur Arudino to SX1276 wire
connection.

Do a things, try to write a code only for DIO0,1,2 pins with LED blinking
and see it led is blinking.

If Led is blinking then Ur naming to DIO0,1,2 is correct or else check pin
naming.

ramesh493, 2017-02-17:
@sabas1080 The error is on the serial port.

i have checked all the pins now its working fine.

Now i am getting  "Starting Packet queue"  On serial port.

what more need to check, do i need anything more to change in the code

sabas1080, 2017-02-17:
This error is usually because the chip is not being detected, check your pins and pin configuration

closed 2017-02-20 by ramesh493