matthijskooijman/arduino-lmic #69: Support Class C

sabas1080, 2017-02-23:
Hi 

Is there any plan to support class C?

Thanks for your job

Oliv4945, 2017-02-26:
I have not tested it but you can try to put your node in `RX_ON` state thanks to `os_radio(RADIO_RXON);` after each reception. See the `raw.ino` example

halukmy, 2018-06-12:
@sabas1080  did you handle class C in lmic? is it works?

sabas1080, 2018-06-14:
@halukmy no

halukmy, 2018-06-14:
@sabas1080 how you handle class-C type messages?

halukmy, 2018-06-16:
@Oliv4945  any update?

sabas1080, 2018-06-24:
Hi @halukmy we made a private solution for class C, with another library

closed 2018-06-24 by sabas1080

gizmocuz, 2018-06-25:
@sabas1080 , please share the name/link of the library

halukmy, 2018-06-25:
@sabas1080  yes please share it

jwktje, 2018-07-02:
I would really want to know this too @halukmy

halukmy, 2018-07-02:
@jwktje  everyone want it, but damn @sabas1080  not hear us.. he just star some noobie repos

NicolasVI2, 2021-09-10:
For LMIC class C support add this function in LMIC.c : 

void setupClassCRx(void){
	
	LMIC.osjob.func = FUNC_ADDR(processRxMessage);
	LMIC.txrxFlags = TXRX_DNW2; 
	LMIC.rps = dndr2rps(LMIC.dn2Dr);
	LMIC.freq = LMIC.dn2Freq;  // frequency the same as RX2 window
	LMIC.dataLen = 0;
	os_radio(RADIO_RXON);
	
}

Call this function on each event "EV_TXCOMPLETE"

nor24o, 2021-09-29:
> For LMIC class C support add this function in LMIC.c :
>
> void setupClassCRx(void){
>
> ```
> LMIC.osjob.func = FUNC_ADDR(processRxMessage);
> LMIC.txrxFlags = TXRX_DNW2;
> LMIC.rps = dndr2rps(LMIC.dn2Dr); [...]

hi can you tell me how is "processRxMessage " implemented  thx

NicolasVI2, 2021-09-29:
Hi,
void processRxMessage (xref2osjob_t osjob) { if( LMIC.dataLen != 0 ) { LMIC.txrxFlags = TXRX_PING; if( decodeFrame() ) { reportEvent(EV_RXCOMPLETE); return; }   setupClassCRx(); } // Pick next ping slot engineUpdate();}
A call to "setupClassCRx" was added in processRxMessage because some time a bad message is received and we need to relaunch rx listening.

nor24o, 2021-09-29:
Thank you for the fast reply 
I managed to add the code you provided to lmic.c respectively in lmic.h and sketch, but for me, something is not ok , I manage to join and send the first message but after that, I get the following error in the serial console :

> 16:31:43.069 -> 5771987: Ready for uplink
16:31:43.069 -> 5772152: Updating info for TX at 5768835, airtime will be 5792. Setting available time for band 1 to 6348035
16:31:43.069 -> FAILURE 
16:31:43.104 -> C:\Users\horva\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:523

NicolasVI2, 2021-09-29:
Your problem is not related to the function added for class C support but is globaly an uplink problem related to another problem ? But I can't help you because I use LMIC library in a C projet and not in an arduino one.

MehdiV96, 2022-07-09:
> For LMIC class C support add this function in LMIC.c :
>
> void setupClassCRx(void){
>
> ```
> LMIC.osjob.func = FUNC_ADDR(processRxMessage);
> LMIC.txrxFlags = TXRX_DNW2;
> LMIC.rps = dndr2rps(LMIC.dn2Dr); [...]

Hello, I want to setup C class with LMIC library and stm32L0 micro, I have problem to setup it, can you help me via email?, I really need your help, thank you

NicolasVI2, 2022-07-09:
Hello,
It s a custom electronic  board ?  If yes, I can't help you because the hardware part can't be debug via email. If you have still problem you can send me your hardware and I can setup lorawan for you at low rate.
Best regards,
Nicolas

jwktje, 2023-07-09:
Yeah 5 years later I’m still wondering too. No sarcasm

matthijskooijman, 2023-07-10:
I'm not longer maintaining this library, see the README for more info and a recommendation of an alternative library.

halukmy, 2023-07-10:
> Yeah 5 years later I’m still wondering too. No sarcasm

@jwktje  chatgpt told me this this trick, is that possible?

To enable Class C using Arduino and LMIC (LoRaWAN-MAC-in-C), you can follow the steps below:

Download the LMIC Library: Open the Arduino IDE and select "Library Manager." Search for the LMIC library and install it.

Enable Class C Support: The LMIC library is initially configured for Class A (low power) devices. However, Class C (continuous operation) support needs to be enabled. Open the lmic_project_config.h file.

Locate the section where lmic.lmic_band = 915E6; is specified and add the following code below it:

cpp
Copy code
#define DISABLE_PING
#define LMIC_ENABLE_DeviceTimeReq 0
#define LMIC_ENABLE_MultiChannels 1
Edit Your Arduino Sketch: You may need to make the following changes in your Arduino sketch:

Replace lmic.txChnl with LMIC.txChnl.
Replace lmic.devaddr with LMIC.devaddr.
Replace lmic.seqnoUp with LMIC.seqnoUp.
Replace lmic.seqnoDn with LMIC.seqnoDn.
Class C Configuration: To enable Class C usage, add the following code at the beginning of your Arduino sketch:

cpp
Copy code
LMIC_setClockError(MAX_CLOCK_ERROR * 1 / 100);
LMIC_setClass(C);
By following these steps, you can enable Class C using Arduino and LMIC. However, it's important to review the code carefully and customize it as needed since the requirements may vary for each project.

trlafleur, 2023-07-10:
You may want to look at Seeed Studio E5 module, it’s a complete radio and
stack in one easy to use package, fully support LoRaWan, serial interface…

halukmy, 2023-07-13:
> You may want to look at Seeed Studio E5 module, it’s a complete radio and stack in one easy to use package, fully support LoRaWan, serial interface…
> -- ~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~ Tom Lafleur

supports arduino and esp32?

trlafleur, 2023-07-13:
it uses AT command, and supports almost all processors....

~~ _/) ~~~~ _/) ~~~~ _/) ~~~~ _/) ~~

Tom Lafleur

halukmy, 2023-08-01:
> 

can you share a arduino code please

conxwae, 2023-09-29:
Hi all
I share some additional code to make LMIC support CLASS_C which I have been doing, I use PRO MINI 3v3 and a gateway with 8 Channels, the results so far are good for uplink (confirmed/unconfirmed) / downlink (unconfirmed) in ABP or OTAA mode.
Does not support multicasing and downlink (confirmed) has not been successful.
Hope this helps.. cheers

### `config.h`
#### Add the sketch in the `config.h` file 
```c
// Added by Conxwae #######################
// Class C
#define CLASS_C
// End Conxwae ############################
```

### `lmic.h`
#### Find and add the sketch in the `lmic.h` file
```c
enum 
    {
        RADIO_RST = 0,
        RADIO_TX = 1,
        RADIO_RX = 2,
        RADIO_RXON = 3,
        // Added by Conxwae - CLASS_C ########################
        RADIO_RXC = 4,
        // End Conxwae - CLASS_C #############################
    };
```

### `lmic.c`
#### Find and add the sketch in the `lmic.c` file 

```c
// TX/RX transaction support

static void setupRx2(void)
{
    ...
}

// Add by Conxwae - CLASS-C ####################
#if defined CLASS_C
static void setupRxC(void)
{
    LMIC.txrxFlags = TXRX_DNW2;
    LMIC.rps = dndr2rps(LMIC.dn2Dr);
    LMIC.freq = LMIC.dn2Freq;
    LMIC.dataLen = 0;
    os_radio(RADIO_RXC);
}
#endif
// End Conxwae - CLASS-C #######################
```

```c
// ======================================== Data frames

// Fwd decl.
static bit_t processDnData(void);

// Added by Conxwae - CLASS-C ############################
#if defined CLASS_C
static void processRxClassC(xref2osjob_t osjob)
{
    if (LMIC.dataLen != 0)
    {
        // LMIC.txrxFlags = RX_CLASS_C;
        if (decodeFrame())
        {
            reportEvent(EV_RXCOMPLETE);
            setupRxC();
            return;
        }
    } // Pick next ping slot engineUpdate();
}

static void startRxClassC()
{
    LMIC.osjob.func = FUNC_ADDR(processRxClassC);
    setupRxC();
}
#endif
// End Conxwae - CLASS-C #################################

```
```c
static void processRx2DnData(xref2osjob_t osjob)
{
  
  '''
  
    processDnData();
    // Added by Conxwae - CLASS-C #################################
#if defined CLASS_C
    startRxClassC();
#endif
    // End Conxwae - CLASS-C #################################
}
```
### `radio.c`
#### Find and add the sketch in the `radio.c` file 

```c
// start transmitter (buf=LMIC.frame, len=LMIC.dataLen)
static void starttx()
{
// added by Conxwae - CLASS_C ############################
#if defined CLASS_C
    ASSERT((readReg(RegOpMode) & OPMODE_RXC) != 0);
#else
    ASSERT((readReg(RegOpMode) & OPMODE_MASK) == OPMODE_SLEEP);
#endif
    // End Conxwae - CLASS_C #############################
    if (getSf(LMIC.rps) == FSK)

    ....
```
```c
enum
{
    RXMODE_SINGLE,
    RXMODE_SCAN,
    // Added by Conxwae - CLASS_C ##############################
    RXMODE_CLASS_C,
    // End Conxwae - CLASS_C ##############################
    RXMODE_RSSI,
};
```

```c
static CONST_TABLE(u1_t, rxlorairqmask)[] = {
    [RXMODE_SINGLE] = IRQ_LORA_RXDONE_MASK | IRQ_LORA_RXTOUT_MASK,
    [RXMODE_SCAN] = IRQ_LORA_RXDONE_MASK,
    // Added by Conxwae - CLASS_C ##############################
    [RXMODE_CLASS_C] = IRQ_LORA_RXDONE_MASK,
    // End Conxwae - CLASS_C ##############################
    [RXMODE_RSSI] = 0x00,
};
```

```c
// Add by Conxwae - CLASS_C ##########################
#if defined CLASS_C
                    rxmode == RXMODE_SINGLE ? "RXMODE_SINGLE" : (rxmode == RXMODE_CLASS_C ? "RXMODE_CLASS_C" : "UNKNOWN_RX"),
#else
                    rxmode == RXMODE_SINGLE ? "RXMODE_SINGLE" : (rxmode == RXMODE_SCAN ? "RXMODE_SCAN" : "UNKNOWN_RX"),
#endif
// End Conxwae - CLASS_C ##############################
```
#### Look for all `OPMODE_SLEEP` modes and Add `OPMODE_RXC` mode to keep RXC always active
```c
// Added by Conxwae - CLASS_C #######################################
#ifdef CLASS_C
    opmode(OPMODE_RXC);
#else
    opmode(OPMODE_SLEEP);
#endif
// End Conxwae - CLASS_C ##############################
```

```c
void os_radio(u1_t mode)
{
    hal_disableIRQs();
    switch (mode)
    {
    case RADIO_RST:
        // Added by Conxwae - CLASS_C #######################################
#ifdef CLASS_C
        opmode(OPMODE_RXC);
#else
        // put radio to sleep
        opmode(OPMODE_SLEEP);
#endif
        // End Conxwae - CLASS_C ############################################
        break;

    ...

    // Added by Conxwae - CLASS_C #############################
    case RADIO_RXC:
        // receive CLASS_C frame
        startrx(RXMODE_CLASS_C); // buf=LMIC.frame
        break;
        // End by Conxwae - CLASS_C ###########################
    }
    hal_enableIRQs();
}
```

Sorry if my answer is too long and takes up this thread.. /\

halukmy, 2023-10-12:
can be library update or create a new repo ? 

thanks by the way alot ! @conxwae