matthijskooijman/arduino-lmic #71: Support NiceRF SX1276?

SunRioy, 2017-03-02:
Hi, sir. May I know the library is supporting NiceRF SX1276?Because I had tried NiceRF SX1276 (end node)to communicate with( raspberry pi+ NiceRF SX1276) gateway to TTN but TTN didn't receive any information from end node.

Oliv4945, 2017-03-02:
Hi SunRioy.
It should work as NiceRF board is based on SX1276 chip. Do you have any error ?
Are you sure of your pin configuration ?

A SX1276 can only do one frequency, so did you configured the same frequency for both node & gateway ?

SunRioy, 2017-03-02:
Hi, Olive4945. 
Thank you for quick reply. I am using 915MHz actually, I just wonder it is going to work too?

SunRioy, 2017-03-22:
I found out that Yes it is supported