matthijskooijman/arduino-lmic #72: Tx seem to work on arduino pro mini + xl1276 but my Rpi3+Dragino GW does not get the message

jponzo, 2017-03-03:
Hi,

I`ve implemented a RPI3+Dragino lora gateway which has been successfully tested with another RPI3+Dragino as a sender node.

Now I`ve implemented a smaller and more power efficient node with arduino+xl1276 but I was unable to make a successful transmission.

As per the arduino serial monitor, everything seem to be working fine but can not see any message at the gw side.

Starting
RXMODE_RSSI
Started
338: TXMODE, freq=868100000, len=4, SF=9, BW=125, CR=4/5, IH=0
TX
4184: RXMODE_SCAN, freq=868100000, SF=9, BW=125, CR=4/5, IH=0
RX
144629: TXMODE, freq=868100000, len=4, SF=9, BW=125, CR=4/5, IH=0

Frequency was set to 868.0 Mhz at both sides. Am I missing something? Is there any way I can test that my arduino pro mini node us actually transmitting?

Gateway: 
RPI3 + Dragino Lora/GPS
    rf95.setModemConfig(RH_RF95::Bw125Cr45Sf128);
    rf95.setFrequency(868.1); /* MHz */

Node:
Arduino pro mini + XL1276
  LMIC.freq = 868100000;
const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 8,
    .dio = {4, 5, 7},
};

Thanks in advance,

matthijskooijman, 2017-03-03:
> Frequency was set to 868.0 Mhz at both sides. 

Your code and debug output say 868.1Mhz, is that a typo in your issue? Are all settings the same (SF, BW, CR, Implicit header)? Is the inversion correct on the gateway (there is this thing where the imaginary and real parts of the encoding are inverted at some point so nodes cannot hear each other and gateways cannot hear each other, perhaps you have that configured incorrectly somewhere)?

jponzo, 2017-03-31:
Hi, That was a Typo as you said. I think the problem might be the SF. I have to change it to a non-predefined value

leandroreox, 2017-04-02:
Hi , just to follow up to the @jponzo  comment, now we are transimitting ok but on the gateway that uses this code: https://github.com/tftelkamp/single_chan_pkt_fwd we are seeing that the data is someway encripted and we cannot get it printed flat, any advices here ?