matthijskooijman/arduino-lmic #73: SX1272 not recognized ? (issue on ASSERT(v == 0x22))

CecileDuthoit, 2017-03-03:
Hello !
I am trying to use the lib on Arduino Pro Mini (3.3V 8MHz version), and I have a problem with line https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/radio.c#L691, on the assert. I printed it and the register is 0x00... I saw many posts on similar issues, and it appears that my SX1272 chip may not be recognized, which can be caused by a misconfiguration of the pins.

I checked my lmic_pins many (many) times, and I don't understand what could be wrong. I also checked the power, SX1272 chip VCC pin is around 3.3V so it should be ok. 

I also changed the frequency (on this line https://github.com/matthijskooijman/arduino-lmic/blob/master/src/hal/hal.cpp#L77) to 4, 2 and even 1 MHz, just in case, but it doesnt solve the issue.

I've been working on it for two days, and it did work once ! (not sure of what it sent or not but my code was running well). Just once over hundreds unfortunatelly, but it shows that the chip is ok.

Reading hallard's great comment on this post: https://github.com/matthijskooijman/arduino-lmic/issues/40, I think I've tried everything he said, and I'm out of ideas... Do you have any idea of what I could check/change/try to make it work properly?

Thanks!

gizmocuz, 2017-03-03:
If you are using the things network then:

Are you using OTAA or ABP ?
In case of ABP, your frame counter should always increment by one, also when you restart your node, it should start with +1 of the last send counter.
In the things network application settings for your ABP, you can enable 'Relax frame count'
then you do not have to do this anymore...
It got me crazy the first time

CecileDuthoit, 2017-03-03:
Thanks for your reply!

Actually I don't use TTN, I'm just trying to send raw messages for now, I adapted the raw example for what I want to do, but the problem occurs at initialization, when os_init(); is called (which calls radio_init(), which contains this assert that fails).

I've tried with both my code and raw example, and same issue.... the assert fails. I am not sure to understand what makes this register to 0x22, but it does not.... 

So I guess it is not the same problem that you had with ABP.

CecileDuthoit, 2017-03-07:
Actually I think it's related to the Arduino Pro Mini I'm using. Just tried with Arduino Mega and it works. In readme you specify that your lib works with Atmega328P and the Pro Mini I use only has an Atmega328. If anyone has an idea to make the radio chip recognized by a Pro Mini....

bnzeen, 2017-03-11:
I'm facing the same problem and with pro mini 3.3v 8MHz , have you solved the problem ?

CecileDuthoit, 2017-03-14:
Hello bnzeen! I'm glad you fixed your problem with Pro Mini. Don't you have any other problems then? I didn't manage to solve the one I have with this assert yet unfortunately, but if you succeed in launching LMIC TX/RX using Pro Mini on ATmega328, it means I probably made an hardware mistake.

CecileDuthoit, 2017-03-15:
@bnzeen , could u please give me your pin mapping between your Pro Mini and SX1272?
Mine is : 

SX1272 ---- Pro mini 8MHz 3.3V

antenna ----  not connected
GND ---- GND
VCC ---- VCC 
RX & TX_switch ---- not connected
DI0  ---- D7
DI1  ---- D8
DI2  ---- not connected (I only use LoRa)
NSS ---- D10
SDI/MOSI ---- D11
SDO/MISO  ---- D12
CLK  ----  D13
RESET ---- D9

And still can't assert v=0x22 unfortunately... (my LoRa chip works perfectly using Arduino Mega)
Thanks a lot!

CecileDuthoit, 2017-03-21:
Problem solved, just a contact failure on several pins. The lib works with Pro Mini like a charm !

closed 2017-03-21 by CecileDuthoit