matthijskooijman/arduino-lmic #76: Sleeping in os_runloop()

Roberto6969, 2017-03-20:
Is it possible to put Arduino to sleep in TX_INTERVAL? I would like to put Arduino to sleep between sending messages but i have no idea how to combine linrary sleep function (i.e. LowPower) with LMIC run-time scheduler function.

Oliv4945, 2017-03-20:
An easy way is to sleep the required amount of time you need instead of scheduling the next TX in `TX_COMPLETE` event. So you can replace [this line](https://github.com/matthijskooijman/arduino-lmic/blob/master/examples/ttn-otaa/ttn-otaa.ino#L115) by 
```
      for (int i=0; i<37; i++) {
        // Use library from https://github.com/rocketscream/Low-Power
        LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF);
      }
      do_send(&sendjob);
```

Roberto6969, 2017-10-13:
That was good idea until i change Activation method from ABP to OTAA. Now it sends 4 times than i always receive following error:
FAILURE 
C:\Users\Robert\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:523

Do you have any idea how to resolve this error?

Aayk, 2018-01-17:
How can I do for sleep the 32u4 forever and wakes up when I need to send a LoRa package?