matthijskooijman/arduino-lmic #78: plz help us rectify this error!! we are geeting this error when we compile lmic and ha files!!

Preetham288, 2017-03-27:
Arduino: 1.8.2 (Windows 8.1), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Preetham\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10802 -build-path C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802 -warnings=none -build-cache C:\Users\Preetham\AppData\Local\Temp\arduino_cache_222068 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Preetham\AppData\Local\Temp\arduino_modified_sketch_299158\HelloWorld.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Preetham\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10802 -build-path C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802 -warnings=none -build-cache C:\Users\Preetham\AppData\Local\Temp\arduino_cache_222068 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Preetham\AppData\Local\Temp\arduino_modified_sketch_299158\HelloWorld.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\sketch\HelloWorld.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\lmic" "C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\sketch\HelloWorld.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\lmic" "-IC:\Program Files (x86)\Arduino\libraries\hall" "C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\sketch\HelloWorld.ino.cpp" -o "nul"
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\libraries\lmic\lmic.c
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\libraries\lmic\oslmic.c
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\libraries\lmic\radio.c
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\libraries\hall\hall.cpp
Using cached library dependencies for file: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src\SPI.cpp
Generating function prototypes...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\lmic" "-IC:\Program Files (x86)\Arduino\libraries\hall" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\sketch\HelloWorld.ino.cpp" -o "C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\preproc\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\variants\standard" "-IC:\Program Files (x86)\Arduino\libraries\lmic" "-IC:\Program Files (x86)\Arduino\libraries\hall" "-IC:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI\src" "C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\sketch\HelloWorld.ino.cpp" -o "C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\sketch\HelloWorld.ino.cpp.o"
Compiling libraries...
Compiling library "lmic"
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\libraries\lmic\lmic.c.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\libraries\lmic\oslmic.c.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\libraries\lmic\radio.c.o
Compiling library "hall"
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\libraries\hall\hall.cpp.o
Compiling library "SPI"
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\libraries\SPI\SPI.cpp.o
Compiling core...
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\wiring_pulse.S.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\WInterrupts.c.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\hooks.c.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\wiring.c.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\wiring_analog.c.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\wiring_digital.c.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\wiring_pulse.c.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\wiring_shift.c.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\CDC.cpp.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\HardwareSerial.cpp.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\HardwareSerial0.cpp.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\HardwareSerial1.cpp.o
Using previously compiled file: C:\Users\Preetham\AppData\Local\Temp\arduino_build_400802\core\HardwareSerial2.cpp.o
[... cut ...]
C:\Program Files (x86)\Arduino\libraries\lmic/lmic.c:188: undefined reference to `AESKEY'

C:\Program Files (x86)\Arduino\libraries\lmic/lmic.c:189: undefined reference to `os_aes'

collect2.exe: error: ld returned 1 exit status

Using library lmic in folder: C:\Program Files (x86)\Arduino\libraries\lmic (legacy)
Using library hall in folder: C:\Program Files (x86)\Arduino\libraries\hall (legacy)
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
exit status 1
Error compiling for board Arduino/Genuino Uno.

matthijskooijman, 2017-03-28:
I'm totally unsure why this fails. How did you install this library? What example sketch are you compiling exactly? This library does not have a Hello World example.

Preetham288, 2017-03-28:
We included the library from zip file!! What else changes do we have to do?

matthijskooijman, 2017-03-28:
What zip file? Where did you download it? Please be complete in your answers, I don't have unlimited time handling requests like these.

Preetham288, 2017-03-28:
I downloaded the zip file from git hub !! Cut pasted the library files to
arduino/libraries
It had some error to detect #include math\math.h
But somehow got it figured it out!! Later on when we tried to compile we
came across many errors
Such as
- lmic pin map not in scope
-LD status 1 error error compiling
Lmic unused pin not decraled in scope!!
All error were replated to the header files itself!!
We r still not able to rectify the errors!!
When we tried to add the libraries through include  library add.zip file !!
It said this file doesn't have any library file!! We  have to interface
gateway too since the code is complilng we cannot proceed further!! Need
your help sir!!

gizmocuz, 2017-03-28:
Two days ago i started on a clean system
Downloaded/Installed the latest Arduino IDE (1.8.2)
Made a blink example application->works

Next i downloaded this git repo, installed it in Documents\Arduino\libraries\arduino-lmic\

Opened an example (ttn-otaa), compiled, working

There is nothing wrong here...

matthijskooijman, 2017-03-28:
@Preetham288, it sounds like your setup is severely broken somehow. You've also still not answered my questions about what example sketch you are using exactly, so I won't be able to help you out here, and I'm closing this ticket. Feel free to leave more comments, perhaps other users will want to comment still, but I probably will not.

I would suggest you try a clean install of the Arduino IDE, without any other libraries in a clean sketchbook and see if that helps.

closed 2017-03-28 by matthijskooijman

Preetham288, 2017-03-28:
Hello sir,
Thank you very much we compiled the code