matthijskooijman/arduino-lmic #82: Lower power consumption

Roberto6969, 2017-03-31:
Is it possible to lower power consumption of RFM95? Data says in Sleep mode it consumes 0.2 uA and in Idle mode it consumes 1.5 uA. In my case it consumes 160 uA (ATMEL328P without RFM95 consumes 6 uA). Could be LMIC_shutdown the way or there are any othr ways?

matthijskooijman, 2017-03-31:
I've seen the power usage from the RFM95 well below 10uA (perhaps only 0.2uA, haven't measured that precise), so I suspect there is something in your setup that leaks current. Where did you measure this 160uA? In the RFM VCC line? Did you also measure GND? If that draws less, you're likely leaking current through the I/O pins somewhere.

Roberto6969, 2017-04-01:
@matthijskooijman 
[image]
You have right. The main consumer is BMP280 sensor - even it's power is removed. Without sensor is power consumption about 20uA, what is absolutely great!
I know it is out of this scope but if someone knows how to "shutdown" BMP280 i will be greatful for info. 
And yes: be careful with carefull with those Chinse GY-BM 280 with 6 pins - if they had slightly smaller BMP280 chip then it is fake and does not work (left one on the picture).

jcwren, 2017-04-03:
If your BMP280 is consuming power when power is removed, it is likely you have an I/O line that is set high, and is backfeeding the module. For instance, if you're using I2C to interface to it, SDA and SCL idle high, and have pull-up resistors. To get absolute minimum power consumption, you'd need to disable the I2C interface, switch SDA and SCL to GPIO (instead of connected to the I2C module), and drive them low. Additionally, you'd have to be able to disable the pull-up resistor (typically through a high-side FET), so you're not sinking current through them.

If you want to try to put the BMP280 into lower power mode (about 2.7ua), it looks like writing bits 0 and 1 in register 0xf4 should do that (or so sayeth the data sheet).

If you're using the SPI interface, you may not have any external pull-up resistors, so setting the SPI lines to GPIO and driving them all low should be sufficient.

Roberto6969, 2017-04-03:
Wouldn't it be easier to use CSB pin on BMP280? As it is stated "If VDDIO is supplied, but VDD is not, the interface pins are kept at a high-Z level." But on the other hand manual also says "A direct connection between CSB and Vddio is required". As i understand the CHIP ENABLE function is implemented not through CHIP SELECT (pin 2) but through connecting power to Vdd (pin 8) only. BTW: i using I2C int.

jcwren, 2017-04-03:
Yes, you can switch Vdd as long as you keep Vddio present. Unfortunately, the datasheet doesn't clearly say  what the current draw is when Vdd is low and Vddio is high. If it's the same as putting the part in sleep mode (0.3ua), then your life would be easier by putting the part to sleep rather than adding additional logic to manage Vdd.

Roberto6969, 2017-04-03:
Yes, you are right - meanwhile i have already tested your solution. It is really much easier and power consumption is about 25 uA. The easiest way is to use Sparkfun library https://github.com/sparkfun/SparkFun_BME280_Arduino_Library and to us mySensor.settings.runMode = 1; . BTW i still have no idea what is the difference between Forced Mode 1 or 2.

hallard, 2017-04-08:
@Roberto6969 
I know it's off topic, but what are this nice golden board ? Love them ;)

Roberto6969, 2017-04-09:
@hallard BMP 280 I2C breadboard https://www.bosch-sensortec.com/bst/products/all_products/bmp280 - good replacement for DHTxx sensors

hallard, 2017-04-09:
@Roberto6969 
Thanks, I have some of these BMP280 boards grabbed from ebay, but this red and gold in this post where did you get it ? May be a custom one ?

Roberto6969, 2017-04-10:
@hallard Here is the source https://www.aliexpress.com/item/BME280-Digital-Sensor-Temperature-Humidity-Barometric-Pressure-Sensor-New/32672562548.html?spm=2114.13010608.0.0.VVVh4V

I have tested this modules from this seller and works fine. But be careful and read my previous post - there are modules on the market with fake (smaller) sensor chip as this one: https://www.aliexpress.com/item/I2C-SPI-BMP280-3-3-Digital-Barometric-Pressure-Altitude-Sensor-High-Precision-Atmospheric-Module-for-arduino/32681667220.html?spm=2114.13010608.0.0.5JViOA.

rmoscetti, 2017-07-10:
I am sorry for the off topic.
@Roberto6969 could you please contact me via email?
Thank you in advance

Roberto6969, 2017-07-10:
Sure – here i am

rmoscetti, 2017-07-11:
@Roberto6969 
My name is Roberto too (from Italy). I am working on a weather station based on the Adafruit Feather 32u4 RFM95 LoRa Radio - 868 or 915 MHz ( https://www.adafruit.com/product/3078 ).
The weather station must work with Li battery and no solar panel.
I experienced some issues trying to use the sleep mode and I am wondering how did you obtained such low power consumption with your setup.

Thank you in advance.

Best Regards,
Roberto

Roberto6969, 2017-07-12:
Hi Roberto, i am out of office and i will respond you in a few days. BTW: if you want to achieve the lowest consumption you have to use only CPU with a few passive components. LDO is particulary critical.

I will explain you more when i will ne back.

Regards, Robert

Roberto6969, 2017-07-18:
Hi Roberto,

as i pointed out before if you want to achieve low power consumption as it is possible you have to avoid using development boards. With MCU AT328P with minimal number of components (for example https://electronics.stackexchange.com/questions/53713/what-is-the-minimal-set-of-parts-for-a-circut-with-this-avr-microcontroller).

The second fact is you have to use power down (for example LowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF)) of the MCU.

The third fact is you have to run MCU on lower voltage (3.3V) and lower clock (if it is possible).

There is some another hints but if you take into account this three facts power consumption should drop below 50 uA in sleep mode.

Regards,

Roberto

[https://cdn.sstatic.net/Sites/electronics/img/apple-touch-icon@2.png?v=7b89fddaa66b]<https://electronics.stackexchange.com/questions/53713/what-is-the-minimal-set-of-parts-for-a-circut-with-this-avr-microcontroller>

atmega - What is the minimal set of parts for a circut ...<https://electronics.stackexchange.com/questions/53713/what-is-the-minimal-set-of-parts-for-a-circut-with-this-avr-microcontroller>
electronics.stackexchange.com
My Arduino Uno has a ATMEGA328P-PU microcontroller, and bunch of other stuff on the board. I'd like to program the chip without the Arduino software, and only the ...

Mibeus, 2019-09-17:
Hello, 
I start with same project litte later as you :). I have configuration Atmega pro mini 3,3V 8Mhz, (I removed LDO, LED, LED resistors), BMP280 and RFM95. When I measured in low power mode (lowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF) ) only procesor without any other modules and codes for them, consumption was 4uA. After programing MCU with software for comunication with RFM and BMP in low power mode I had consumption 700uA. After googlin I found solution for sleep BMP and consumption decrease to 65uA. But still is so much. I try measure separatly BMP (0,2uA), separatly RFM (0,3uA), but together with procesor have consumption 70uA. Today I test measure consumption without i2c comunication. But still 65uA.
Looks like consumer is SPI or IO pins on Lora module. 

jcwren wrote :If you're using the SPI interface, you may not have any external pull-up resistors, so setting the SPI lines to GPIO and driving them all low should be sufficient.

Do you mean only set all SPI pins on LOW? 
...
pinMode (13, output);  //same for other pins
digitalWrite(13, LOW);
lowPower.powerDown(SLEEP_8S, ADC_OFF, BOD_OFF)
...
This is correct or I must again some inicialized SPI?

And second question what exactli do IO0 and IO1 on RFM module? during sleep must be low or High?

Thank you.