matthijskooijman/arduino-lmic #83: TX fails below ~2.7 V and makes Arduino reset

CecileDuthoit, 2017-04-03:
Hello!

I am using your lib to make a low power device, based on Pro Mini 3.3V 8MHz and I'm facing an issue with TX at "low" voltage. 
I know the SX1272 chip I have can make it with min 1.8 V (cf. datasheet), and I was expecting Arduino board to fail before but it is actually the contrary:

My Pro Mini is working well at 2.2 V (even with the default 8MHz and initial fuses settings) but TX starts failing around 2.5 V (and of course below 2.5 V) for raw communications (my sketch is inspired by raw.ino example) and around 2.7-2.8 V for OTAA (idem with OTAA.ino example). The symptom is that the Arduino resets before sending, i.e. at this line for RAW communications : 
`  os_radio(RADIO_TX);` 

And in the same time, raw RX is working well even at 2.2 V. 
Any idea of what could be problematic with TX at low voltage ? I don't know if it's related to the lib, to the chip, to the Arduino, or some other mysterious parameter...

Thank you very much!

matthijskooijman, 2017-04-03:
Sounds like your power supply or battery is insufficient. When TX starts, the current usage increases, which typically causes a drop in voltage from the power supply or battery. When using an under-spec power supply or battery or low battery, this drop can be bigger and thus problematic.

CecileDuthoit, 2017-04-04:
OK, thanks for your reply! 
But it is still weird because I tested it using a GBF that can deliver as much current as needed and it still reset around 2.27 V (for RAW TX). 
And it would mean that sending OTAA joining uses significantly more power than sending raw messages?

CecileDuthoit, 2017-04-04:
Actually I measured TX current consumption for both RAW and OTAA scenarii, and the peaks are lower for OTAA (around 80 mA) than for RAW (around 102 mA), so the fact that OTAA version fails "before" RAW version seems weird, don't you think?

hallard, 2017-07-04:
That strange because unless you changed fuses on your Arduino Pro Mini, your Brown Out Detection (BOD) should be set to 2.7V (every Mini I had in hand had this setting) and below, trigger a CPU Reset.
This is why I change my Pro Mini Bootloader with BOD set to 1.8V and using Optiboot to win 1.5K of flash (great with LMIC) I also setup the bootloading to 250KBPS to speed up uploads
I wrote some posts about this for another project called ULPNode
See Bootloader [Part 1](https://hallard.me/ulpnode-bootloader/) and [Part 2](https://hallard.me/ulpnode-bootloader-1/)
But if you Mini works at 2.2V (no BOD ? or 1.8V) may be @matthijskooijman has right and you can try setting a big capacitor (1000uF) between VCC and GND to see it it's better in term os stability ?

BTW, Can you post the result of this reading fuses [sketch](https://github.com/WestfW/fusebytes) ?