matthijskooijman/arduino-lmic #84: Error on MSP430F5529LP

ahmed9207, 2017-04-04:
I'm using [MSP430F5529LP](http://www.ti.com/tool/msp-exp430f5529lp) with [RFM9x](https://learn.adafruit.com/adafruit-rfm69hcw-and-rfm96-rfm95-rfm98-lora-packet-padio-breakouts/overview), when i compile i get the following errors:

C:\Users\Egm001\Documents\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\libraries\arduino-lmic-master\src\hal\hal.cpp:77:14: error: 'SPISettings' does not name a type

C:\Users\Egm001\Documents\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\libraries\arduino-lmic-master\src\hal\hal.cpp: In function 'void hal_pin_nss(u1_t)':

C:\Users\Egm001\Documents\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\libraries\arduino-lmic-master\src\hal\hal.cpp:85:13: error: 'class SPIClass' has no member named 'beginTransaction'

C:\Users\Egm001\Documents\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\libraries\arduino-lmic-master\src\hal\hal.cpp:85:30: error: 'settings' was not declared in this scope

C:\Users\Egm001\Documents\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\libraries\arduino-lmic-master\src\hal\hal.cpp:87:13: error: 'class SPIClass' has no member named 'endTransaction'

C:\Users\Egm001\Documents\energia-1.6.10E18-windows\energia-1.6.10E18\hardware\energia\msp430\libraries\arduino-lmic-master\src\hal\hal.cpp: In function 'u4_t hal_ticks()':

C:\Users\Egm001\Documents\energia-1.6.10E18-windows\energia 1.6.10E18\hardware\energia\msp430\libraries\arduino-lmic-master\src\hal\hal.cpp:154:115: error: 'static_assert' was not declared in this scope

any ideas how to solve this?

Oliv4945, 2017-04-04:
Hi,
It seems that Energia is not using the same functions than Arduino for SPI, you should try to change them for the functions listes on[ this page](http://energia.nu/reference/spi/).

ahmed9207, 2017-04-05:
thanks , i'll try it

ahmed9207, 2017-04-27:
i changed the SPI functions in hal.cpp to this 
```
static void hal_spi_init () {
    SPI.begin();
}

void hal_pin_nss (u1_t val) {
    if (!val)
  //      SPI.beginTransaction(settings);
    {
        SPI.setClockDivider(SPI_CLOCK_DIV4);
        SPI.setBitOrder(MSBFIRST);
        SPI.setDataMode(SPI_MODE0);  
        SPI.begin();
        

    }    
    else
       // SPI.endTransaction();
        SPI.end();
    //Serial.println(val?">>":"<<");
    digitalWrite(lmic_pins.nss, val);
}

// perform SPI transaction with radio
u1_t hal_spi (u1_t out) {
    u1_t res = SPI.transfer(out);
/*
    Serial.print(">");
    Serial.print(out, HEX);
    Serial.print("<");
    Serial.println(res, HEX);
  */  
    return res;
}
```
Now i can join the Network  server however i'm not able to send any data to the gateway.
Any ideas? thanks