matthijskooijman/arduino-lmic #88: Join is attempted on disabled channels

dlarue, 2017-04-12:
after LMIC_disableChannel(x) is called for say channels 0-9 and again for channels 11-72 I still see join attempts on all of the US-915 subband2 frequencies.

christopherdro, 2018-03-20:
@dlarue I'm running across the same issue. Any updates on your end?

kierenblack, 2018-09-28:
I have the same problem. Why is it joining on channels I disabled?

dlarue, 2018-09-28:
This fork seemed to be more actively developed - can you try it and see if it has the same issue and report back here?  

https://github.com/mcci-catena/arduino-lmic

christopherdro, 2018-09-28:
MCCI's library uses LMIC 1.6 and doesn't have this issue.

The bug with this lib is that it doesn't take the disabled channels into account on rejoin attempts.

I used this (https://github.com/async-la/arduino-lmic/commit/9f29f6f8cedd77a6490361990536e51bedad1d5a) hack to force the use of TTN's sub-band, but the better option would be to use library mentioned above.