matthijskooijman/arduino-lmic #89: Cortex M3 Fails to Compile

hewittwill, 2017-04-14:
Trying to use the following module: [http://wiki.in-circuit.de/index.php5?title=radino32_SX1272](url), which required a downgrade to Arduino 1.6.0 to use the compiler it required.

However this puts it below the 1.6.6 minimum for arduino-lmic, if I try to add the c++11 flag to the platforms.txt, it fails to compile as 'Cortex M3 is not valid for --params' 

Any suggestions?

matthijskooijman, 2017-04-14:
I have no clue about this M3 platform and the things needed for that, so I can't come up with anything off-hand. Your report is missing some details, such as links to the exact core/platform.txt you are using, the exact changes you made to add c++11, or the full verbose compiler output, which would help to get an idea of what's happening.

matthijskooijman, 2017-04-14:
Also, you messed up the markdown for the link, you had the link text and link url reversed. I fixed that for you.

hewittwill, 2017-04-14:
See below for the error report WITHOUT attempting to change to a C++11 compiler

```
Using library IBM LMIC framework in folder: /Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master 

/private/var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/AppTranslocation/F42A92CD-796D-4FDE-B3A8-D4B0C08F8825/d/Arduino.app/Contents/Resources/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_radino32 -DVECT_TAB_BASE -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -mcpu=cortex-m3 -DF_CPU=24000000L -DARDUINO=10601 -DARDUINO_RADINO32 -DARDUINO_ARCH_STM32L1 -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32L1__ -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libopenstm32l1 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/include -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/stm32f1/include -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/usb/stm32f1 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/usb/usb_lib -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/cores/radino32 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/variants/radino32 -I/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src /var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/build7153785696326729011.tmp/sketch_apr14c.cpp -o /var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/build7153785696326729011.tmp/sketch_apr14c.cpp.o 
/private/var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/AppTranslocation/F42A92CD-796D-4FDE-B3A8-D4B0C08F8825/d/Arduino.app/Contents/Resources/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/arm-none-eabi-gcc -c -g -Os -w -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DBOARD_radino32 -DVECT_TAB_BASE -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -mcpu=cortex-m3 -DF_CPU=24000000L -DARDUINO=10601 -DARDUINO_RADINO32 -DARDUINO_ARCH_STM32L1 -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32L1__ -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libopenstm32l1 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/include -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/stm32f1/include -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/usb/stm32f1 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/usb/usb_lib -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/cores/radino32 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/variants/radino32 -I/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src /Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src/aes/lmic.c -o /var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/build7153785696326729011.tmp/IBM LMIC framework/aes/lmic.c.o 
/private/var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/AppTranslocation/F42A92CD-796D-4FDE-B3A8-D4B0C08F8825/d/Arduino.app/Contents/Resources/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/arm-none-eabi-gcc -c -g -Os -w -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -DBOARD_radino32 -DVECT_TAB_BASE -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -mcpu=cortex-m3 -DF_CPU=24000000L -DARDUINO=10601 -DARDUINO_RADINO32 -DARDUINO_ARCH_STM32L1 -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32L1__ -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libopenstm32l1 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/include -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/stm32f1/include -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/usb/stm32f1 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/usb/usb_lib -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/cores/radino32 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/variants/radino32 -I/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src /Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src/aes/other.c -o /var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/build7153785696326729011.tmp/IBM LMIC framework/aes/other.c.o 
/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src/aes/other.c: In function 'os_aes_cmac':
/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src/aes/other.c:64:9: error: 'for' loop initial declarations are only allowed in C99 mode
         for (u1_t i = 0; i < 16; ++i, ++buf, --len) {
         ^
/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src/aes/other.c:64:9: note: use option -std=c99 or -std=gnu99 to compile your code
/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src/aes/other.c:99:13: error: 'for' loop initial declarations are only allowed in C99 mode
             for (u1_t i = 0; i < sizeof(final_key); ++i)
             ^
[... cut ...]
     uint32_t scaled = micros() >> US_PER_OSTICK_EXPONENT;
                              ^
/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src/hal/hal.cpp:154:115: error: 'static_assert' was not declared in this scope
     static_assert(US_PER_OSTICK_EXPONENT > 0 && US_PER_OSTICK_EXPONENT < 8, "Invalid US_PER_OSTICK_EXPONENT value");
                                                                                                                   ^
Error compiling.
```

Perhaps to do with the version of the Arduino IDE being older?

matthijskooijman, 2017-04-14:
Seems you added c11 support, but not c++11. Try adding `-std=gnu++11` (IIRC) to `compiler.cpp.flags`.

hewittwill, 2017-04-14:
Solved every single issue but the fact that the micros() function is still unavailable.

I tested it just by setting the relevant variable equal to millis(), which is definitely supported, and it compiled fine. Are there any alternative functions to micros?

See the error report:

```
Arduino: 1.6.1 (Mac OS X), Board: "Radino32 USB-Load"

Using library IBM LMIC framework in folder: /Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master 
Using library SPI in folder: /Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/libraries/SPI (legacy)

/private/var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/AppTranslocation/5D768DED-CD0D-4CFE-B30F-5BEAA2FD3DA5/d/Arduino.app/Contents/Resources/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/arm-none-eabi-g++ -std=gnu++11 -c -g -Os -w -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_radino32 -DVECT_TAB_BASE -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -mcpu=cortex-m3 -DF_CPU=24000000L -DARDUINO=10601 -DARDUINO_RADINO32 -DARDUINO_ARCH_STM32L1 -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32L1__ -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libopenstm32l1 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/include -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/stm32f1/include -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/usb/stm32f1 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/usb/usb_lib -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/cores/radino32 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/variants/radino32 -I/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/libraries/SPI /var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/build1609351609630272981.tmp/ttn-abp.cpp -o /var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/build1609351609630272981.tmp/ttn-abp.cpp.o 
Using previously compiled file: /var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/build1609351609630272981.tmp/IBM LMIC framework/aes/lmic.c.o
Using previously compiled file: /var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/build1609351609630272981.tmp/IBM LMIC framework/aes/other.c.o
Using previously compiled file: /var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/build1609351609630272981.tmp/IBM LMIC framework/aes/ideetron/AES-128_V10.cpp.o
/private/var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/AppTranslocation/5D768DED-CD0D-4CFE-B30F-5BEAA2FD3DA5/d/Arduino.app/Contents/Resources/Java/hardware/tools/gcc-arm-none-eabi-4.8.3-2014q1/bin/arm-none-eabi-g++ -std=gnu++11 -c -g -Os -w -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_radino32 -DVECT_TAB_BASE -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -mcpu=cortex-m3 -DF_CPU=24000000L -DARDUINO=10601 -DARDUINO_RADINO32 -DARDUINO_ARCH_STM32L1 -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32L1__ -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libopenstm32l1 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/include -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/stm32f1/include -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/usb/stm32f1 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/system/libmaple/usb/usb_lib -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/cores/radino32 -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/variants/radino32 -I/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src -I/Users/hewittwill/Documents/Arduino/hardware/ICT_Boards/stm32l1/libraries/SPI /Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src/hal/hal.cpp -o /var/folders/wc/dfpc_z2j779517nsdnj0l2_w0000gn/T/build1609351609630272981.tmp/IBM LMIC framework/hal/hal.cpp.o 
/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src/hal/hal.cpp: In function 'u4_t hal_ticks()':
/Users/hewittwill/Documents/Arduino/libraries/arduino-lmic-master/src/hal/hal.cpp:136:30: error: 'micros' was not declared in this scope
     uint32_t scaled = micros() >> US_PER_OSTICK_EXPONENT;
                              ^
Error compiling.
```

Oliv4945, 2017-04-14:
>I tested it just by setting the relevant variable equal to millis(), which is definitely supported, and it compiled fine. Are there any alternative functions to micros?
On the STM32L151 you can use the [SysTick](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0552a/Babieigh.html).

Also I am a bit surprised by the flag "-DMCU_STM32F103CB" as the module embed an STM32L151
You should consider to use Semtech or mBed stacks, they have been developed on you uC.

hewittwill, 2017-04-14:
Cheers @Oliv4945 

I've yet to test it, but the plan thus far is to use the Timers to create a manual micros() function and implement that in the library.

I was *hoping* I'd be able to use this module and then not have to learn how to use the mBed platform, as I haven't had any experience with it, so we shall see!

rk125, 2017-04-19:
Hi,
I was having same errors trying to compile for STM32F103C board with original stm32duino and arduino-lmic. Try this good guy's work from here https://github.com/tomtor/Arduino_STM32 and https://github.com/tomtor/arduino-lmic - after changing compile is good so far.