matthijskooijman/arduino-lmic #93: EV_JOINING block

N4zbr0cK, 2017-04-27:
Hello,
I saw quite a lot of tutorials, advices, DIY, to solve my problem, but I still can't reach the solution.

I have an Adafruit 32u4, wired IO1 to 6, pin mapping modified as follow:
const lmic_pinmap lmic_pins = {
    .nss = 8,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 4,
    .dio = {7, 6, LMIC_UNUSED_PIN},
};

A "delay(1000);" added at the beginning of the setup, changed clk to 8E6 instead of 10E6, modified DR for join procedure from DR7 to DR9 (one or another doesn't change anything), added the clock error modification line as mentionned quite a few times on forums...

I did quite a lot of things, but it looks like the hardware still wants to be stronger than me.
I am quite new in IoT, so I wonder what I am doing wrong.

Anyone of you has got any other idea ?

Already connected this board to a gateway ?

Thanks for your advices.

j4zzcat, 2018-04-12:
I'm seeing the same issue on the LoRa32u4. Joining packet do get to the ttn network, an accept join packet is sent back to the device, but apparently never received, thus is stays in EV_JOINING.

genoinosonda, 2018-06-15:
@j4zzcat  I'm also having the same problem.
Could you tell me how to solve it?

j4zzcat, 2018-06-15:
See solution here: https://www.thethingsnetwork.org/forum/t/lora32u4ii-ev-txcomplete-not-called/13431