matthijskooijman/arduino-lmic #94: loss of USB recognition with Arduino M0

EtienneBourdeau, 2017-04-28:
Hello ! 

I'm using an Arduino M0 (ARM Atmel SAMD21 microcontroller) with a SX1272 LoRa module.
But when I transfer your code on the Arduino, after having set up config.h and the keys in the 'ttn-abp' example, the Arduino isn't recognized by Windows anymore (Windows 7 64b pro). I can't find it in the Device Manger either, and nothing seems to work.

Can you help me plz ? 
Maybe I didn't understand something.
Thank you very much !

EtienneBourdeau, 2017-05-09:
After a few research, I found out that the problem comes from the use of the "hal_disableIRQ()" function in "os_runloop_once", but I don't know how to debug it.
Could someone help me ?

Thank you once again !

PhatHub, 2017-06-02:
I'm not an M0 expert, but looking at the Adafruit [FAQ for their M0 Feather](https://learn.adafruit.com/adafruit-feather-m0-basic-proto/faq?view=all#faq) there seems to be an entry about crashed sketches that causes an [error with starting the bootloader.](https://learn.adafruit.com/adafruit-feather-m0-basic-proto/faq?view=all#faq-6)

At that point, the solution might be hitting the reset button two or three times during the early phase of uploading a sketch?

selvamaniraj, 2017-06-06:
Hello all, 
      I am also using arduino samd controller trying to upload the program i have also same problem( EtienneBourdeau ). if anyone find  a solution.help me to upload the program.