matthijskooijman/arduino-lmic #97: 328P (Nano) only one transmit in ABP example

PhatHub, 2017-05-31:
I'm running an Arduino Nano v3 clone (with HL-340 UART) and I'm interfacing the RFM95w to it with a TXS0108 level shifter. I'm programming it with the ABP example sketch.

I can get the Arduino Nano to transmit one packet (I can see it received in my gateway), but after that the Nano crashes and resets. Linux then reassigns the HL340 with a different ttyUSB number. (USB1 to USB0, and vice versa)

Has anyone else encountered something similar?

jcwren, 2017-05-31:
Refer to #91 and see if that might not be what you're experiencing.

matthijskooijman, 2017-06-01:
If the ttyUSB number changes, that means the USB chip also resets. I would suspect problems in your power supply, are you sure it can deliver sufficient current? Perhaps the voltage sags during transmission resetting the USB chip and MCU but not the transceiver, so it can finish its transmission?

PhatHub, 2017-06-01:
**jcwren:** Thanks, but I had already set DIO2 to unused since I'm only dealing with CSS. Very interesting though!
**matthijskooijman:** I'll have to check that! HL/CH-340 version is not very well documented in terms of those operating limits, but according to someone's "reverse" effort, it may have a 20-30ma limit on the (internally regulated by the CH/HL-340) 3.3v rail:
[Arduino Nano CH340 – schematics and details](http://actrl.cz/blog/index.php/2016/arduino-nano-ch340-schematics-and-details/)
That is a far lower limit compared to the original Nano v3, which the external 3.3v regulator theoretically can handle up to 150ma.
I guess that's what I get for going cheap! I'll try to add a nice 3.3v regulator to my design... wish me luck :)

closed 2017-06-01 by PhatHub

matthijskooijman, 2017-06-01:
> it may have a 20-30ma limit on the (internally regulated by the CH/HL-340) 3.3v rail:

I don't have numbers handy, but I'm pretty sure 30mA is not enough, so that's likely your problem indeed :-)

Good luck!

PhatHub, 2017-06-02:
Thank you! Whew! It was a long day, but basically I found a way to solder a surface mount AMS1117 crooked onto the 3.3v and GND lines... and I attached a wire to tap 5v rail. That solved the UART resetting. (I believe the RFM95w requires about 100mA at most...)

Then I found out that the DIO0/1 pins were incorrect... I was going by 328p pin numbers instead of Arduino-mapped pin numbers... (those beautiful Arduino pin-out posters are a bit confusing between the gray and purple numbers) and that solved an issue where my sketch couldn't detect that there was a EV_TXCOMPLETE! 

Now, it's working beautifully.

matthijskooijman, 2017-06-06:
Cool, thanks for reporting back :-) Have fun!