matthijskooijman/arduino-lmic #98: 433Mhz frequency

ELIVB, 2017-06-01:
Hello
I am quite new in Arduino and LoraWan,  the Lmic library support only  915Mhz and 868Mhz, what about 433Mhz?

Thanks

eli

PhatHub, 2017-11-04:
Did you try setting LMIC.freq=433000000?

PhatHub, 2017-11-29:
I'm sorry... looking back at my response, I was thinking in RAW mode, not lorawan/MAC.
I've been also looking into the 433 standard, but I haven't found enough information published yet to see how the lorawan standard is set for it. By comparison, eu868 and us915 are both fairly well laid out by the Lora Alliance.
So far I've tried: 
- I tried duplicating the eu868-specific code for the most part (or added to most preprocessors by using `defined(CFG_eu868) || defined(CFG_xx433)`).
- I'm not sure how the channels were spaced, so I just filled up the 8 channels with 200khz center freq spacing. These channels were implemented in `lmic/lorabase.h`. 
  - At least the first three channels were specified in the Lora Alliance's "LoraWan 2016 Regional Parameters": `433.175, 433.375, 433.575` That may be all you need? 
  - Also interesting is Multitech's default channels in their `global_conf.eu433.json` where they use `433.2, 433.4, 433.6,` and `433.8` which is a little off from Lora Alliance's regional parameters...?
- The LMIC_setupChannel in `lmic/lmic.c` will have to be edited too for the appropriate frequencies to dBm (power?) correlation values. 
- The last thing I couldn't figure out was what the DR_PAGE_XX### (in this case, DR_PAGE_EU433) was about in `lorabase.h`

I got it to compile, but I haven't tested it yet. Also, my remaining Arduino is a Mega2560, which I have issues trying to get operational with the LMIC's rtos.

As far as I've seen, there aren't very many 433mhz gateways... although I see that brocaar's loraserver has the  EU433 specs worked out last year. Maybe a few months from now I'll eventually purchase a 433 gateway to play with.

MauriPastorini, 2018-01-02:
I have the same question, Can I use this library to comunicate 433mhz lora sx1278 modules? What do I have to change for it?
If someone could answer me please!
Best regards

chris-gunawardena, 2018-02-18:
Have you tried this? http://wiki.dragino.com/index.php?title=Connect_to_TTN#How_To_add_other_frequency_band_such_as_EU433

gclairec, 2018-02-23:
Hi, Does anybody have any fix on this? My device is stuck in TX_COMPLETE. No packets were received in TTN

the-louie, 2018-06-08:
I made a naivë adaption of the library to add eu433 support. I added eu433 sections everywhere I found a eu868 section and copied over the values I could find regarding 433 and left all other values the same as for 868. Most of the information comes from http://wiki.dragino.com/index.php?title=Connect_to_TTN#How_To_add_other_frequency_band_such_as_EU433

I have two 433MHz RAK831 gateways configured to talk to a loraserver.io server and it works! 😄 

The code is available here: https://github.com/the-louie/arduino-lmic/tree/louie-add-naive-eu433-support

gizmocuz, 2018-06-08:
@the-louie , if you look at this document, on what frequency are you allowed to transmit in your country ? What's the bandplan ?

the-louie, 2018-06-08:
@gizmocuz In my country 433-435 MHz falls well within the range of allowed frequencies to use without license.

was1822, 2022-01-18:
Starting
FAILURE 
C:\Users\SAP3\Documents\Arduino\libraries\arduino-lmic-master\src\lmic\radio.c:689
i am using 1278 lora node with dragino gateway LG 308 at 433MHZ i am facing this ssue