matthijskooijman/arduino-lmic #100: packet size and compilation issue

rmoscetti, 2017-06-09:
Dear all,
I am working on a project for the development of a weather station based on arduino.
I had a strange issue when the size of the packet to send is higher than 52 bytes.

Example of code that properly work:
[code]
byte packet[52];
LMIC_setTxData2(1, packet, sizeof(packet), 0);
[/code]

Example of code that does not work:
[code]
byte packet[53];
LMIC_setTxData2(1, packet, sizeof(packet), 0);
[/code]
When the packet size exceeds the 52 bytes, the compiled code was also affected by another strange issue. In fact, its size was about the half of the compiled sketch that properly work.

How is it possible? Is there a limit in the packet size that the LMIC library can handle?

Thank you in advance for your help.

Best Regards,
Roberto

matthijskooijman, 2017-06-11:
I believe 52 bytes is the maximum payload size that LMIC has allocated buffers for, see https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/lorabase.h#L183

rmoscetti, 2017-06-11:
Thank you. I will split my packet in two to circumvent this problem.

rmoscetti, 2017-06-20:
Just an update.
Because my project is a weather station, I was able to use the delta econding to reduce the packet size to 49 bytes.
https://en.wikipedia.org/wiki/Delta_encoding

agusakovs, 2017-08-22:
I was able to make it send more than 52 bytes ( in my case I've tested it with 115 bytes). All I had to do is swap the 64 value to 128 in the https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/lorabase.h#L40 and https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/lmic.h#L30
Hope that helps.

sommmen, 2017-12-19:
Note: the lmic's framesize needs to be a _multitude of 8 bits_ (so, 32 - 64 - 128 - 256 etc.).

Please note: that KPN only allows for a max of 64 bytes anyways, so bumping the framesize up sometimes (in my case) doesn't matter at all.

See:
https://zakelijkforum.kpn.com/lora-forum-16/frequently-asked-questions-faq-8329

avbentem, 2018-05-12:
@sommmen, while the KPN FAQ indeed says:

> **What is the size of KPN LoRa packets?**
> The maximimum size that can be used all the time is 51 bytes. (When FOpts field within the LoRa packet is empty, this can be 54 bytes).

...I wonder about _"that can be used all the time"_. Maybe the FAQ is just referring to SF12? (Which makes sense for most use cases, as if a device needs to work at SF12, it does not make sense to have it send more data at better data rates, even if a provider would allow for that.)

I'm not using KPN, but if you have any definitive answer then I'm surely curious.

asanz060, 2021-03-30:
> I was able to make it send more than 52 bytes ( in my case I've tested it with 115 bytes). All I had to do is swap the 64 value to 128 in the [...]

Hi, I have tried to change these 2 values but I still have the problem of not being able to send more than 51 bytes. Is there anything else I should change?