matthijskooijman/arduino-lmic #104: regRead : 0

b-rohit, 2017-07-11:
I am following [this](http://www.mobilefish.com/developer/lorawan/lorawan_quickguide_build_lora_node_rfm95_arduino_uno.html) tutorial to make LoRa node with arduino node. I uploaded the example sketch(ttn-abp) successfully. I got following error > 
```
Starting
FAILURE 
/home/rohit/Arduino/libraries/arduino-lmic-master/src/lmic/radio.c:689
```
It is an assertion error. I put debug messages to read registers. I found that all the register's value is zero. Could someone point out what am I doing wrong

matthijskooijman, 2017-07-11:
Only reading zeroes often indicates a broken SPI connection. Check your wiring, or try lowering the SPI speed.

b-rohit, 2017-07-12:
@matthijskooijman Thanks. soldering on chip wasn't done properly. I checked the connection and it's up and working.

closed 2017-07-12 by b-rohit