matthijskooijman/arduino-lmic #105: Problem Sending data use Pro mini and RFM95

VanXungUIT, 2017-07-14:
Hi!

I configured pin with Node.

const lmic_pinmap lmic_pins = {
    .nss = 10,
    .rxtx = LMIC_UNUSED_PIN,
    .rst = 9,
    .dio = {2,5, 7}, //DIO0 and DIO1 connected
};

And When I run code

Starting
Packet queued
214: EV_JOINING

And I waiting, but not change.
Can anyone explain to me? And how to solve it?

sabas1080, 2017-07-14:
Hi @VanXungUIT 
What frequency are you using?
Activate in config.h

LMIC_DEBUG_LEVEL 2

LMIC_PRINTF_TO Serial
To know what is happening

VanXungUIT, 2017-07-15:
Hi @sabas1080 
I configured
#define CFG_eu868 1

#define CFG_sx1276_radio 1
When I edited file config.h 
#define LMIC_DEBUG_LEVEL 2
Result:

9442737: TXMODE, freq=868100000, len=23, SF=8, BW=125, CR=4/5, IH=0
9760401: RXMODE_SINGLE, freq=868100000, SF=8, BW=125, CR=4/5, IH=0
9825781: RXMODE_SINGLE, freq=869525000, SF=12, BW=125, CR=4/5, IH=0
9835802: engineUpdate, opmode=0xc

VanXungUIT, 2017-07-15:
@sabas1080  I waiting and not print data to uart.

I think it just print EV_JOIN_FAILED and EV_TXCOMPLETE. But it not.

VanXungUIT, 2017-07-15:
I using Gateway Esp8266 Signal Channel and frequency 868.1 Mhz

matthijskooijman, 2017-07-15:
JOIN_FAILED only happens after all retries are done (at all SF levels), so that could take dozens of minutes or even a few hours.

You say you use a single channel gateway - did you take measures to ensure that LMIC transmits on the right frequency with the right settings? Can you see the packets coming in on the gateway?

Are you using TTN? I recall that single-channel gateways are not used for downlink packets (since they cannot support all settings or something like that). If that's correct, single channel gateways cannot be used for OTAA joining, so perhaps that's your issue.

navinyate, 2017-07-15:
Are u using 3.3v or 5v version.  5v is not capable of OTAA it can be
achieved in 3.3v version.  ABP can be achieved in both

matthijskooijman, 2017-07-15:
Huh? The 3.3V version of what? The Pro Mini? Why would that be? I'm running a 3.3V 328-based board with OTAA just fine (in fact, it's easier to connect, since the RFM board needs 3.3V I/O, not 5V). Or are you referring to the ESP / gateway?

navinyate, 2017-07-15:
No,  3.3v of Pro mini works well with OTAA but not 5v version I ment

VanXungUIT, 2017-07-15:
@matthijskooijman  I running a 3.3V 328. 
Yes I using TTN. I do not see packet coming in on the gateway. 
I waiting 4 hours but It not change

I building gateway esp8266 link: https://github.com/things4u/ESP-1ch-Gateway-v4.0.

matthijskooijman, 2017-07-15:
> No,  3.3v of Pro mini works well with OTAA but not 5v version I ment

And *why* do you think this is? Using a 5V pro mini requires more work (level translators), but it should work just fine. If you leave out the translators, I can't see why ABP would work but not OTAA, though perhaps RX stops working if you overvoltage the RFM but TX doesn't care or something?