matthijskooijman/arduino-lmic #108: How can I debug the library

pierrot10, 2017-07-23:
Hello.
I am very new and I am trying to send "hello word" to TTN. with that librar and with Adafruit Fether Mo Lora board.

It seam to partialy work and I reevie that message

> Packet queued
> 28436586: EV_TXCOMPLETE (includes waiting for RX windows)
> Packet queued
> 29749481: EV_TXCOMPLETE (includes waiting for RX windows)
> Packet queued

I can not really undestand what does it means. As far I understand, the message is queued and the sending process is completed. But In my TTN console I do not sue a record.
(I make sure to enter the correct Devise addres, the Network session key and the App session key into the sketch file (ttn-adp.ino))

Why nothing i recorded at TTN?

My main worrie is how to debug the loop(), particulalery the function
`os_runloop_once();`

I found the function here
[https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/oslmic.c#L92](url)
I tried to add a Serial.println() but it did not compile.

Then I look at config file and I uncomment that line
[](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/config.h#L33)

Then into the function os_runloop_once()
I added a Serial.println() and then a printf("hello"). Serial.print() generate an error while I complie and printf() did not print anythings.

I also observed when I remove Serial.print() and prinf() and I kept that line uncomment
[](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/config.h#L33)
i received that error messgae
C:\Users\pierrot\Documents\Arduino\libraries\arduino-lmic-master\src\hal\hal.cpp: In function 'void hal_printf_init()':

> C:\Users\pierrot\Documents\Arduino\libraries\arduino-lmic-master\src\hal\hal.cpp:223:54: error: '_FDEV_SETUP_WRITE' was not declared in this scope
> 
>      fdev_setup_stream (&uartout, uart_putchar, NULL, _FDEV_SETUP_WRITE);
> 

Then I wundering how I can debug into the file
[](https://github.com/matthijskooijman/arduino-lmic/blob/master/src/lmic/oslmic.c)
How can I print in my terminal the value of some variable?

Why when I uncomment #define LMIC_PRINTF_TO Serial, I got an error?

And by the way, could you recommend me some step to use your library with an Adafruit Feather MO Lora board?

Many thank for helping me to start with LORAWAN and TTN.

matthijskooijman, 2017-07-24:
> I can not really undestand what does it means. As far I understand, the message is queued and the sending process is completed. But In my TTN console I do not sue a record.
(I make sure to enter the correct Devise addres, the Network session key and the App session key into the sketch file (ttn-adp.ino))

That's correct. Are you really sure you entered the right addresses? Did you enter the devaddr LSB-first and the keys MSB-first?

> '_FDEV_SETUP_WRITE' was not declared in this scope

This suggests you are not using an AVR-based Arduino. The LMIC_PRINTF_TO currently only works on AVR. This is on a Adafruit Feather M0? That would make sense.

Could you test if this commit fixes the error message and makes printf work? https://github.com/matthijskooijman/arduino-lmic/commit/69cdef5eaf3ad51cf666609af317b2f651c03c82

Beliwars, 2017-08-09:
Hello @matthijskooijman  I tested commit https://github.com/matthijskooijman/arduino-lmic/commit/69cdef5eaf3ad51cf666609af317b2f651c03c82 but it didn't work for my case, I'm working with a Nucleo-64 (STM32L053R8) using the STM32 Arduino core. My guess is that as for esp8266 the STM32 arduino core doens't implements the fopencookie; this is the error I've got when trying to compile.

..\Documents\Arduino\libraries\arduino-lmic-master\src\hal\hal.cpp:237:8: error: 'cookie_io_functions_t' does not name a type

 static cookie_io_functions_t functions = {

        ^~~~~~~~~~~~~~~~~~~~~
..\Documents\Arduino\libraries\arduino-lmic-master\src\hal\hal.cpp: In function 'void hal_printf_init()':

..\Documents\Arduino\libraries\arduino-lmic-master\src\hal\hal.cpp:245:37: error: 'functions' was not declared in this scope

     stdout = fopencookie(NULL, "w", functions);

                                     ^~~~~~~~~
..\Documents\Arduino\libraries\arduino-lmic-master\src\hal\hal.cpp:245:46: error: 'fopencookie' was not declared in this scope

     stdout = fopencookie(NULL, "w", functions);

                                              ^

exit status 1
Error compiling for board Nucleo-64.

I was trying to debug because the library is not working with my HW setup; on the Serial console I only got "Start" and I don't see anything else; I checked the connections reading the register on the SX1212 using basic SPI read operations and they are ok.

Beliwars, 2017-08-09:
FYI, I managed to get the library working with the STM32 Arduino core; the execution was stopping after the noInterrupts() command on the hal_disableIRQs function from hal.cpp; I don;t know yet why I guess this core is using Interrupts some place else and it doesn't like to get all interrupts disabled; anyway after commenting this line I could join a GW and send the message "Hello World"on the ttna-otaa example.

@pierrot10 I could debug using the lmic_printf macro which is sending messages to the Serial port on my STM32 board; I found this while trying to activate the LMIC_PRINTF_TO Serial; I activated LMIC_DEBUG_LEVEL and started to see debug messages on the Serial monitor.

faehrlich, 2017-08-09:
@pierrot10 Hi. I am using the Adafruit Feather M0 with RFM95. I am trying the same thing (Hello World) and get the same Output in the serial monitor:

```
378921: EV_TXCOMPLETE (includes waiting for RX windows)
Packet queued
4464875: EV_TXCOMPLETE (includes waiting for RX windows)
Packet queued
8641583: EV_TXCOMPLETE (includes waiting for RX windows)
```

Do you was able to fix this or was the node able to join in any way ? My keys are correct and were checked a lot of times. 

 @matthijskooijman : Do you have any other suggestion? My keys are correct and were checked a lot of times.

jerrylogansquare, 2017-08-18:
I'm getting the same 'EV_TXCOMPLETE' - 'Packet queued' sequence.  I'm using ESP8266 with Adafruit LoRa (HopeRF RFM9x).  I setup my device in the senetio (USA- Chicago) portal (see graphic), and copied the arrays of bytes directly into the abp.ino file.  

[image]

pierrot10, 2017-08-28:
Hello, thank for all of your comments. I am actually re-working with that  library and unfortunately, I have the same issue. I can not debug my code, in spite of your suggestion.

@matthijskooijman 
Yes, I am using an Feather MO Lora for Europe
Yes I tried the commit 69cdef5 but it did not fix my problem.

I am actually trying the file ttn-abp.ino. I am in the train and there is no getway aroung my way. When I run the code, it print the following

> Starting
> do_send
> VBat0: 4.20
> VBat1: 4.20
> Packet queued
> Go to loop

Then I am trying to understand what happen with the following functions
```
os_runloop_once();
    //os_runloop();
```
It's the reaon why, I would to be ale to pring some debug message with lmic_printf() (or something else) to see what happen in

```
os_runloop_once();
// end
os_runloop();
```

What do the above function print when there is no gateway and when the train a running trough a city where there are getways (I am going to go trhought a city with gateway in about 20mn).

Thank a lot if I can get some tips with my Fether MO Lora.
Cheers

iotlearner0level, 2018-04-12:
@Beliwars can you share your library pleas i am kind of stuck. it says starting and it think it crashes somewhere...

matthijskooijman, 2020-01-26:
I've just pushed an update that should enable printf on non-AVR architectures too.

Since this is not really a bug, I'm closing this issue.

closed 2020-01-26 by matthijskooijman