matthijskooijman/arduino-lmic #110: Downlink parameter setting?

PhatHub, 2017-08-05:
Currently I'm working with an Arduino, a Mirakonta-like gateway, and the brocaar loraserver.io backend. Currently (according to the UDP traffic) the backend is telling my gateway to downlink with the following parameters: SF10, BS125, CR4/5, ipol true.

How do I set up a datarate to set up the RX parameters on the Arduino?

(On a side note, has anyone successfully gotten a @brocaar loraserver.io package to successfully downlink to an Arduino with this LMiC stack?)

matthijskooijman, 2017-08-05:
RX1 downlink happens with the same settings as uplink. RX2 downlink is specified by the LoRaWAN spec (SF12 IIRC), but can be changed by the server through the join accept message. If the server does not support this, or you're using ABP, you can [set dn2Dr manually](https://github.com/matthijskooijman/arduino-lmic/blob/master/examples/ttn-abp/ttn-abp.ino#L215).

closed 2017-08-05 by matthijskooijman

PhatHub, 2017-08-07:
Is there a way to change the coding rate to 4/5 instead of the default 4/8? I got it to SF10 and BW125.

I saw the line about the dn2Dr, but is there a parameter constructor that lets me create the specific parameters I need?
Thanks!

PhatHub, 2017-08-08:
However, I don't know if the "reverse/inverse polarity" is correct...

PhatHub, 2017-08-08:
OKAY I finally understand now. Just adding here in case someone else runs into the same issue as me...

For some reason my node wasn't reading inverted IQs.
In `lmic/radio.c` there's several lines which indicates to the modem to read inverted IQs:

>  #if !defined(DISABLE_INVERT_IQ_ON_RX)
>      // use inverted I/Q signal (prevent mote-to-mote communication)
>      writeReg(LORARegInvertIQ, readReg(LORARegInvertIQ)|(1<<6));
>  #endif

However, `DISABLE_INVERT_IQ_ON_RX` was apparently defined, probably because I was testing node-to-node connections earlier. So to correct that, I commented out in `lmic/config.h`:

> //#define DISABLE_INVERT_IQ_ON_RX

And now I have RX2 working.

> 38117513: engineUpdate, opmode=0x800
> 38742513: engineUpdate, opmode=0x808
> 38742890: TXMODE, freq=903300000, len=20, SF=8, BW=125, CR=4/8, IH=0, FCNT=0x0000002e
> 38806802: RXMODE_SINGLE, freq=926300000, SF=8, BW=500, CR=4/8, IH=0
> 38861734: RXMODE_SINGLE, freq=923300000, SF=10, BW=125, CR=4/8, IH=0
> 38900338: Received downlink, window=RX2, port=1, ack=0
> 38900384: EV_TXCOMPLETE (includes waiting for RX windows)
> No ACK!
> Received 3 bytes of payload

Whew! Thanks  @matthijskooijman for helping me get the parameters correct.